opentx/opentx #4373: openTx 2.2 RC10 no telemetry in multiprotocol

benoit35690, 2017-02-02:
OpenTx 2.2 RC10
Taranis X9D
multiprotocol module with casing from Banggood flashed with latest firmware version (v1.1.6), Telemetry enabled for Bayang protocol (see configuration delails in _config.h)

In both companion and Taranis, when the Bayang protocol is selected, it is not possible to enable the telemetry option.

[_Config.h.zip](https://github.com/opentx/opentx/files/747172/_Config.h.zip)

3djc, 2017-02-02:
what exactly do you call "enable the telemetry option." ?

3djc, 2017-02-02:
Seems ok here
[image]

benoit35690, 2017-02-02:
in X9D 
- i didn't notice the Status field
- below the multiprotocol line there is a line "No_telemetry...."
I can provide a photo but not before this evening (Paris time) ;-)

I don't understand the meaning of the integer (from 0 to 100) for telemetry. I would expect the feature to be ON or OFF.

3djc, 2017-02-02:
No telemetry means your module as crashed (you can confirm by checking the red led is off when that happens). It used to happen to me a lot, until I remove a lot of protocol I did not need but enable 'just in case', and now it does not happen at all anymore (I'm around 50% usage reported by Arduino)

schwabe, 2017-02-02:
The options are protocol specific, see the Bayang protocol:

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#bayang

The Status field is since RC9. You probably have an older version on your X9D. Or if you have status and it shows NO MULTI_TELEMETRY, see https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Tx-Taranis.md#protcol-status-in-serial-mode-opentx

closed 2017-02-02 by schwabe

benoit35690, 2017-02-02:
OK thank you, i will double check on my side.
Sorry for my lack of knowledge on this topic

benoit35690, 2017-02-02:
I have done some double checkings
- Taranis version is RC10
[image]
- my module is at 95% usage
- I have never seen my module crashing (red led always on, model fly always without lost)
- my module is flashed with 1.1.6 version (see attached config)
[_Config.h.zip](https://github.com/opentx/opentx/files/747692/_Config.h.zip)

1) my Syma x5 is bound and flies
The status is incorrect 

[image]

2) assan protocol is not included in my module protocol
The status is incorrect 
[image]

3) my H101 firmware enables telemetry
The status is incorrect
[H101config.zip](https://github.com/opentx/opentx/files/747716/H101config.zip)

So I thing this issue should be reopen

schwabe, 2017-02-02:
I think the status correct. The status says "No Multi_telemetry" which means that OpenTX does not receive multi telemetry. 

At this point I suspect that OpenTX receives no telemetry at all. Your config enables INVERT_TELEMETRY and MULTI_TELEMETRY so that should be good. Check the DIY wiki for the instruction on enabling telemetry on a hw level for multi. 

If you own a logic analyzer you could capture the telemetry pin to see what is happening there.

benoit35690, 2017-02-02:
- for non supported protocol (assan in my case) shouldn't I get on the screen "Prot. Invalid" ?
- for a supported protocol (syma in my case) shouldn't I get "V1.1.6... bind" ?
 like described in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Tx-Taranis.md#protcol-status-in-serial-mode-opentx

schwabe, 2017-02-02:
TO quote myself (twice):

* from this issue: At this point I suspect that OpenTX receives no telemetry at all. 

* from the section in the DIY manual: If you see NO MULTI_TELEMETRY or no NO TELEMETRY in the setup screen of the external module, like this: you are missing the MULTI_TELEMETRY and/or INVERT_TELEMTRY option in your _Config.h and/or **you need to hardware modifications to enable telemetry**.

benoit35690, 2017-02-02:
according to
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Transmitters.md
only Turnigy 9X or 9XR require HW mod, X9D is said to be compliant with telemetry without mod.
Module does not require modification in all cases.

kilrah, 2017-02-02:
Wrong, module requires modification in all cases (solder the little bridge).

schwabe, 2017-02-02:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Module_BG_4-in-1.md

Also all modules require soldering to have a 6pin ISP Header. Only the newest batch of modules requires no soldering for serial input/telemetry.

benoit35690, 2017-02-02:
Sure, but this is to enable the serial, nothing to do with the telemetry.
If this mod were not made on my module, I would not been able to use the serial mode with the protocols I have tested.
My module is the Version 1.2 and the soldering has been done at Banggood

kilrah, 2017-02-02:
There are 2 bridges, one is for serial and the other for telemetry. You can have one and not the other, so check that.

benoit35690, 2017-02-02:
1 for TX and 1 for RX, yes both bridges have their solderings

schwabe, 2017-02-02:
Then we are at a point where without a logical analyzer or acess to the hardware we have no idea what is going on there.

benoit35690, 2017-02-02:
I don't have an analyser but if you are able to build a debug version that logs traces, I can test it.
Did the feature worked previously with another module ? just to perform a non regression test

kilrah, 2017-02-02:
Current multi code is broken, MULTI_TELEMETRY indeed doesn't work with it. Please report that to them, nothing to do with us.

benoit35690, 2017-02-02:
OK, thank you for this exchange ;-)

kilrah, 2017-02-02:
Correction, merge issue. Works perfectly fine.

schwabe, 2017-02-02:
Since you have a Taranis you probably also have Frsky receivers. You could bind the reciever in the Frsky protcol to the module and see if the telemetry works there. And double check your soldering.

Also your module might just be broken. The code already logs if it receives somthing strange. They be logged to USB-serial if you get a fw with the cli option enabled.

kilrah, 2017-02-03:
If you're not getting telemetry with a FrSky receiver bound in the right mode to the internal module and with the multimodule disconnected the Taranis likely has a hardware issue.

benoit35690, 2017-02-03:
I do get telemetry through the internal module, not through the external multiprotocol

soonik, 2017-02-08:
It seems like I have exactly the same issue as benoit35690. My v1.2 module(on the back it says YRFM02_V1.2) from BG came with the bridges pre-soldered for serial mode. Also uncommented

#define TELEMETRY
#define INVERT_TELEMETRY
#define MULTI_TELEMETRY
#define ENABLE_SERIAL

and all protocol specific telemetry options. But on my XD9+ running OpenTX 2.2.02.2N362, it says NO MULTI_TELEMETRY. I can bind and fly my quad with FlySky X6B RX(AFHDS 2A).

powercycle, 2017-02-08:
The No Multi_telemetry error can occur if you have commented out any of the modules in your _Config.h if using banggood 4-in-1.  See https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/48
Also, most recent multi code is 1.1.6.16

benoit35690, 2017-02-08:
In my _Config.h all modules are enabled.

dev-fred, 2017-02-08:
Did you flash the fuses of the microcontroller: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling.md#flash-the-firmware
It's the other source of crash if not properly configured.

benoit35690, 2017-02-08:
"If you have a 4in1 Multiprotocol module you can skip this step."
My module has never crashed (red led always on and models fligh without signal lost)

dev-fred, 2017-02-08:
Ok, wrong advise,  you have no crash.
However, I have a 4in1 module (and I have also helped a guy in same situation) where the fuse was not those given here https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling.md#flash-the-firmware
Just check that you have efuse=0xFD or 0x05 	hfuse=0xD3 	lfuse=0xFF, 
you can skip this step but just verify !
In particular efuse determines if you protect or not the eeprom during flash.
You loose the binds done each time you flash, if the value is not good.

benoit35690, 2017-02-08:
OK thank you, I will try this when I will be back from business trip