opentx/opentx #4387: RC10 N362: USB / hangs -> on USB disconnect deletes the current Model.

tubeme, 2017-02-03:
I've had this at least 4 times. May be it is due to my SD card which is pretty old one and most likely slow. But anyway it is a nasty bug.

I setup my Horus as a SD mass storage device. So it is recognized in windows as (H:) HORUS device without a problem. But then once in a while when I connect the USB to PC I hear the USB sound but the device is not recognized OK. It is with just (H:) signature without the name of the SD card and also inaccessible. That is pretty normal situation no worry at all, it happens all the time. 

The problem is that at this very moment when I disconnect the Horus from the PC's usb. The Horus deletes the current Model (the Active Profile), and creates a new blank Model with the index number.

The actual model is still on the sd card so I could recover it by entering a line in RADIO/models.txt.

But at first I though it deleted an hour of work...

This happened at least 4 times today. So I'm buying a new SD card tomorrow to see if it is SD card related. But at least the Horus should be aware of such situation and not delete the profiles (or rather not remove them from the lists)...

projectkk2glider, 2017-02-03:
It does this, because it can't see the model files on the SD card. Which means that there is a problem or a bug with SD card access. Probably the same problem prevents you from seeing driver in Windows. 

Please try new SD card and report back.

tubeme, 2017-02-06:
So far no problems with new card.

projectkk2glider, 2017-02-07:
OK, closing for now, reopen if issue reappears with the new SD card.

closed 2017-02-07 by projectkk2glider