opentx/opentx #4393: New Companion UI - LS's in Output window don't update unless the window has focus

jtaylor2, 2017-02-04:
Windows 10 VS-15update3 QC=5.7.0 build of Next Companion for X9D 02/04/17 10:33 EST

The Local Switch's in the Radio Outputs window do not show set unless the window has focus.  When you go back to the Simulator window the Radio Outputs LS's, if set, loose their indication.  This prevents observing the LS state as you change the switches on the simulator.  The Global and Channel displays in the Outputs window continue to show changes when the window looses focus, it is only the LS's that stop updating when focus is lost.

jtaylor2, 2017-02-04:
I just compiled and tested on openSUSE Leap 42.2 and it does not have this problem.  The Logical Switches update even if the Outputs windows does not have focus.  I am compiling with qt 5.6.1 on Linux and 5.7.0 on Windows so don't know if the difference is qt version or Linux/Windows difference.

kilrah, 2017-02-04:
Confirmed on Windows. Docking the outputs window in the main canvas works around the problem.

jtaylor2, 2017-02-04:
I have been unable to get any window to dock.  Maybe I don't know what I'm doing, but when I take it to the main canvas it stacks the windows and gives me tabs to them on the bottom of the main canvas.

kilrah, 2017-02-04:
Either double click on the window title, or first make the main window bigger

mpaperno, 2017-02-04:
OK thanks for report, will investigate Windows LS update issue.  Strange that GVars still work since it's essentially the same code.

BTW, recommend updating to Qt 5.7.1 as 5.7.0 has at least one known bug with dockable windows.

Also, learned the "hard" way... on Windows at least, after updating Qt install, make sure to replace the DLLs which are in your program's running folder with the new ones.  An app built with 5.7.1 will run perfectly fine with 5.7.0 DLLs and you'd never even know it.  It'll only complain if you try to mix in a newer DLL with the older ones (as I discovered when added Qt5Svg.dll).

kilrah, 2017-02-04:
> Strange that GVars still work since it's essentially the same code.

They don't ;)

I'm on 5.7.1, and did update the DLLs.

mpaperno, 2017-02-04:
> > Strange that GVars still work since it's essentially the same code.

>They don't ;)

Ah I was going on:

> The Global and Channel displays in the Outputs window continue to show changes when the window looses focus, it is only the LS's that stop updating when focus is lost.

OK well that would be more logical if neither worked.  Pun intended.  :)

kilrah, 2017-02-04:
Well, let me rewrite that as a helpful bug report..

- Both GVARS and LSs actually do update whether the window has focus or not
- Both GVARs and LSs lose the "color" focus when the window loses focus
- Since GVARs show a number and the active FM in bold you don't actually lose information when the color square disappears
- Since the color square is the only indication an LS is active, LSs become useless when the window loses focus

Maybe the LS buttons could be set as "pressed" as an additional indication...

mpaperno, 2017-02-04:
Ah.... that makes a lot of sense now, thanks!  I thought I was being clever by avoiding use of CSS to indicate the LSs.  I'm using the standard "highlight" palette color for background instead.  Actually having another indicator besides color for LS is a good idea anyway.

commit f82ee65a (2017-02-06): [Simulator] Radio Outputs: Fix wrong FM highlighted (closes #4408) and current FM not highlighted until switched once; Change highlight color role of active LS/FM so it is always shown (fixes #4393); Active LSs now much more obvious (bold font & sunken frame).
commit ded843b9 (2017-02-06): [Simulator] UI tweaks for last major update. (#4414)

* [Simulator] Debug output: Adjust exclusion filter, help text <kbd> background color, fix help text typos & mark as translatable.

* [Simulator] Radio Outputs: Fix wrong FM highlighted (closes #4408) and current FM not highlighted until switched once; Change highlight color role of active LS/FM so it is always shown (fixes #4393); Active LSs now much more obvious (bold font & sunken frame).

* [Simulator] Rework some SimulatorMainWindow functionality:
    * Fix floating dock widgets appearing behind other applications on Linux (https://git [... cut]
closed 2017-02-06 by bsongis