opentx/opentx #4395: Telemetry ratio for BtRx, A1, A2

dev-fred, 2017-02-04:
I create A1 in entering 5V, but I have no way to enter directly the ratio 13.2 in the grey area: 
[image]
But magically, if I kill the simulator window and run it again, the ratio appears:
[image]

What is the expected way to simulate BtRx, A1, A2 ?

Tested on NEXT of today but seen on 2.2.0N362

dev-fred, 2017-02-04:
Add: If I edit A1, the value ratio is already 13.20 but it's not taken in account

kilrah, 2017-02-04:
The ratio field is simply displayed for information, read form the sensor configured in the radio.

closed 2017-02-04 by kilrah

dev-fred, 2017-02-04:
The sequence to get all sensors working is :
1-Open telemetry simu
2-Activate Discover new sensors in radio
3-Enter a value in simu
That's all, the sensor is created and the value is displayed in the radio

For A1, A2 and BtRx we have this sequence:
1-Open telemetry simu
2-Activate Discover new sensors
3-Enter a value in simu
Here the sensor is created and the value displayed in the radio is always = 0
3-Kill the telemetry simu window
4-Open telemetry simu
And this time we have the correct value

This sequence is a workaround to simulate A1 but I am surprise that you think it's a normal sequence.

kilrah, 2017-02-04:
You're supposed to have telemetry already configured in your model, the simulator isn't really meant to create/discover sensors.

dev-fred, 2017-02-04:
It's your assumption but there is no technical reason todo that way.
I have really appreciated to build LUA scripts based on Cels without any HW and this way works fine.
This way can be apply to all sensors excepted for A1, A2.
You have decided to grey the ratio and not to get the default configuration already there.
This workaround (kill/re-run simu) shows simply that we can work with A1/A2 also this way but it's not your choice .
Sorry but it's not very coherent.

kilrah, 2017-02-04:
> there is no technical reason todo that way.

Sure there is - The telemetry simulator does not have the ability to adjust the ratio in the radio.
Then within companion any changes that are done in the simulator are lost, so detecting sensors there makes no sense.

dev-fred, 2017-02-04:
However, I can adjust the ratio in the radio (after the famous kill/re-run) 
[image]
I imagine that ratio variable can be accessed by both side. 

Companion can't help for LUA, it's natural to remain in the simu

dev-fred, 2017-02-04:
In fact, the real problem is not to be able to adjust the ratio, it's already configured at 13.2.
In order to use the telemetry simulator, we are supposed just to enter the voltage and get the same value in the radio (and the grey ratio should be at 13.2 like the radio and not at 0).
At the moment, to have this simple behaviour, we need to kill and re-run the telemetry simulator (which becomes complicated with the new docking system).
There is nothing to save in Companion which is unable to deal with LUA script, and creating sensors is so easy.
I can survive with the workaround I have found, and may be the way I use the simu is a corner case but I don't like your conclusion "so detecting sensors there makes no sense." 
This non sense works very well when dealing with LUA script and sensors from scratch without HW.

kilrah, 2017-02-04:
>At the moment, to have this simple behaviour, we need to kill and re-run the telemetry simulator

Only the very first time when the sensor doesn't exist yet, or if you change the ratio while it runs... Once your sensor is there you don't need to restart.

> (which becomes complicated with the new docking system).

What? It's simpler than ever, just click twice on the telemetry simulator icon.

dev-fred, 2017-02-05:
> What? It's simpler than ever, just click twice on the telemetry simulator icon.

Thanks for this tip, the double-click is much better than kill+re-run. 
My comment was just linked to the way I used the docking system and the visual effect produced  with the automatic resize when you have 3 windows docked and you kill one.
I like very much this new improvment 🥇 

In conclusion the workaround is easier to use with the new docking system and the double-click but idealy, A1,A2 should not behaves differently from other sensors when we enter a value.

kilrah, 2017-02-05:
The problem is they actually DO behave differently technically from all different sensors, and the "reading ratio from the radio" is already a workaround to make it more convenient, otherwise you'd have to enter a raw 0-255 number.

dev-fred, 2017-02-05:
I have also tested the case with a telemetry already configured. 
I enter 5V and 6V and nothing happens in the radio: 
[image]

I need to double-click on the icon telemetry to get value:
[image]

It seems that something is not iniatlized in A1,A2 when we check Simulation:
[image]

kilrah, 2017-02-05:
Can't reproduce, for me if a sensor exists the greyed ratio fields always get populated when the simulator is opened. Please give exact way to reproduce.

dev-fred, 2017-02-05:
OK, I can reproduce: there are 2 ways to enter in simulator.
1- From Edition of the model with the button: 
[image]
in this case, the simulator is directly openned on the model you are editing and the greyed ratio field is populated if  A1 is defined.
2- From this window:
[image]
In this case, if the default model has no sensor and if you navigate with the radio to a model with sensors the greyed ratio field is populated after a double-click.
In the previous post, I have clicked on a model with sensors A1 and A2 defined but I have started the simulation using the way 2. This was not volonteer but this view is not coherent.

kilrah, 2017-02-05:
All normal then. The ratio field is only populated when the window is opened, so if you change model you need to close/reopen to reload the value from the new model.

dev-fred, 2017-02-05:
Yes, all normal.
Finally to avoid the double click, I can simply fill the values in Model Edit/Telemetry
[image]
This time the ratio field is populated when the window is opened without create/discover sensors.