opentx/opentx #4400: Companion Mixer Copy problem, n362

rdeanchurch, 2017-02-04:
In Companion I tried to copy a mix from Ch16 to Channel 10.
QX7 selected as transmitter, Operating system Ubuntu.
[image]

projectkk2glider, 2017-02-05:
Please explain what the actual problem is.

rdeanchurch, 2017-02-05:
In the picture above there are now 2 sets of Channel 7 and 9.  One set  in the correct order and another set after Channel 10.
This also occurred when I tried to add another channel after Channel 9 by double clicking on the empty Channel 10.
There is no channel 11 an 12 now showing after the copy.

bsongis, 2017-02-05:
I cannot reproduce. Unless I completely misunderstood the issue

kilrah, 2017-02-05:
Can't reproduce either.

So if you say "This is repeatable" please give a step-by-step guide to reproduce.

rdeanchurch, 2017-02-05:
Rename the following files to remove the .png.and leave as .otx
With Qx7 selected In Comp.n362, open RdyToCopy.otx
Doubleclick 03_Bixler-
goto the tab mixers.
Select and right click and copy ch16.
highlight ch!0
right click and paste.

In my case the result is like
the file copyDone.otx.

[image]
[image]

kilrah, 2017-02-05:
Yep that breaks indeed, thanks.

bsongis, 2017-02-05:
ouch interesting ... I take it

rdeanchurch, 2017-02-05:
FYI, This was migrated from an X9D ( ! +) 2.0.20 to 2.1X9D to 2.2X9D then to Qx7.
Not sure that helps, but thought it might be relevant.

bsongis, 2017-02-05:
And did you have something in channels CH8 => CH16 before? It seems to me they should not be displayed!

rdeanchurch, 2017-02-05:
Only Ch 9
     MAX -80 SDup
+= MAX -40 SDmid
+= MAX  80 SDdwn
as shown in orig post.
And Ch16  a 0 to 100 representation of Thr output

bsongis, 2017-02-06:
Your file rdytocopy.otx is already corrupted, do you have your real eeprom file from X9D?

bsongis, 2017-02-06:
Here is a way to produce a corrupted .otx file:
* you select a X9D+ profile
* you create a model with a mix using SH as source
* you switch to a X7 profile (SH becomes ???)
* you export this file to the real X7 - or save it to a .bin (you ignore the warning about the ??? source)
* you reimport from X7 or the .bin file  ** => your model is corrupted**

projectkk2glider, 2017-02-06:
His model data is already courupted it contains unused mixer lines:
```
(gdb) set $pos=0
(gdb) p model->mixData[$pos++].destCh 
$18 = 1
(gdb) 
$19 = 1
(gdb) 
$20 = 2
(gdb) 
$21 = 3
(gdb) 
$22 = 4
(gdb) 
$23 = 4
(gdb) 
$24 = 5
(gdb) 
$25 = 5
(gdb) 
$26 = 6
(gdb) 
$27 = 0
(gdb) 
$28 = 7
(gdb) 
$29 = 0
(gdb) 
$30 = 9
(gdb) 
$31 = 9
(gdb) 
$32 = 9
(gdb) 
$33 = 0
(gdb) 
$34 = 16
```

rdeanchurch, 2017-02-06:
You are correct that is the way I made the file.
2.2X9d to 2.2QX7

bsongis, 2017-02-06:
Yes, cf 2 posts above :)

bsongis, 2017-02-06:
@rdeanchurch I see at least 3 problems with this issue, I will come with related fixes

kilrah, 2017-02-06:
There is no way to handle that "correctly". Whatever we do will be wrong, we can avoid further corruption by turning it into a "valid" value but it will introduce more confusion. E.g. your SH will become THR or so, which you won't notice until you observe wrong model behavior.

The real solution is that when you see the warning that "??? doesn't exist on this board" you should stop and correct until there are no more warnings. Maybe we could actually abort the transfer, that way you HAVE to correct it now while you still can recognise the wrong things by them being "???".

rdeanchurch, 2017-02-06:
OK

kilrah, 2017-02-06:
Regardless of that there's still a bunch of bugs to solve and a decision to take about that default behavior!

bsongis, 2017-02-07:
Yes please don't close this issue now.

rdeanchurch, 2017-02-07:
I suppose just greying the LR, RS, SwE, and SwG  and leaving them would make X7 too much of a special case.  A field in the eepe could be set as a count of all instances of those missing inputs, and until the count was reduced to zero, that model would be just give a warning ion tx and simulator.
Or they could be changed to virtual inputs with a value of zero.   vLR, vRS, vSwE, and vSwG.  Probably too simplistic, I'll drop out an let someone else be in charge of the issue'sclosing.

schwabe, 2017-02-07:
I think the data structures for the X7 do not have the same size as the X9D and we have no way of represeting these switches or do we?

PR #4425 (2017-02-10): Bsongis/issue 4400
It should fix #4400 
Please review
Discussion needed about all the TODO I introduced
closed 2017-02-10 by bsongis