opentx/opentx #4401: Q-X7 - RC10(N362) Switch Warning State editing is obscured and intermittent

mythreeleggedcat, 2017-02-05:
RC10 N362 running in the Companion Simulator on Windows 7SP1

I'm assuming the simulator runs the actual firmware in simulation so for the most part works identically to the actual radio...I don't have a Q-X7 so I can't confirm if this problem also exists on the Radio.

Enable/Disable of switch warning in Model Settings does not work consistently.  You scroll to the switch you want, and it is displaying, say '0', and the '0' flashes.  You press Enter and the '0' still flashes, but if you scroll away '-' is revealed indicating you have disable the warning.  If you scroll back to the switch position the  '0' returns.  Pressing Enter again will not necessarily change the underlying state from '-' back to '0'.  Sometimes pressing Enter repeatedly will change the state but you cannot see this unless you scroll away.
[image]
[image]

closed 2017-02-07 by 3djc

mythreeleggedcat, 2017-03-07:
I confirmed this is fix in the simulator and the new enable/disable interface is much clearer.  Haven't tried it on the Q-X7 radio yet.  Will flash that and try it tomorrow.

However, now there is no icon or indication for "Set switches to desired check position" state.  You scroll past the F essentially to the blank space at the end of the line and then it works to set the switches and press the "ENTER" knob !  A bit mysterious....  I think on the X9D+ it highlights all the switches when in this mode which could work here.  A flashing character in that last space would also work as an indicator.  And alternating up then down arrow might be fun.  Or if flashing and alternating is too involved,  just have a ? appear or highlight in that last space when you scroll past the F

3djc, 2017-03-07:
Long enter now works anywhere on that line, much like it does on x9

mythreeleggedcat, 2017-03-08:
OK but there is no visual indication on the Q-X7.  On the X9D+ the line of switches highlights...in 2.2 RC10.  I think it has been that way for a while.

I will flash both later with RC11 and compare.

3djc, 2017-03-08:
Yes there used to be a signed with not much meaning that was removed to clear some space. We tried several options on X7, and this one was thought to work best given the ergonomics of the radio

mythreeleggedcat, 2017-03-08:
Ok just tried it on the X9D+ 2.2 RC11 and when you reach the Switches field the line of switches highlights and you can long press enter to record the current switch state.  
[image]
If you short press enter it goes into individual enable/disable mode with short press enter changing the state.  Long press enter has no effect in this mode.
[image]
probably how it has been for a while.

I see the Q-X7 changes this so it doesn't highlight the whole line first and scroll to the next field but instead scrolls through the individual switches allowing you to edit the enable state with short press enter...this obscures the "record current switch state" long press enter feature but somewhat compensates by making it always available.  

Not worth a battle but honestly seems a bit of a gratuitous change to me.  The X9D+ way would work just as well.  Either way it is a somewhat obscure interface so I'm not sure it benefits anyone to make them different.

I can see reason to do the Q-X7 way if the scroll wheel makes it hard to program scrolling within a field...but otherwise...when there is no reason not to be consistent, consistency is generally preferred...particularly if it is is obscure.

3djc, 2017-03-08:
I plan to revisit this anyway, but I seem to recall the issue was that the fonts where not allowing the full line seletion

PR #4405 (2017-02-07): X7 switch warning setup
With this version, the switch been edited is both blinking and inversed, which seemed most natural to me when editing. Another version (https://github.com/opentx/opentx/tree/3djc/X7-sw-warning) has those blinking for editing, and  blinking+invers for editing activated

This closes #4401