opentx/opentx #4402: Q-X7 Input and Mix editor short cut to Channel Display doesn't work

mythreeleggedcat, 2017-02-05:
RC10(N362) running in Companion Simulator on Windows7SP1

May be a intentional or just not implemented yet but the shortcut using long press Menu (as with the X9D+) when in the Model Editor to see the Channel Monitor does not seem to work.  It still works on the X9D+ both in the simulator and on the Radio

Also configuring the firmware to include "mixermon" does not seem to have any effect.  Pressing Enter when the Channel Monitor is displayed just stays on the Channel monitor.  This also doesn't work in the 2.2 RC10 X9D+ simulator but mixermon does work on that radio with that version.  It works for the X9D+ in the 2.1 simulator.  I had downloaded X9D+ firmware that did not include it and flashed the radio, then downloaded X9D+ firmware that did include it and flashed the radio.  It did not work from a synced X9D+ profile nd otx file or from a new .otx file created with the same profile active.

kilrah, 2017-02-05:
There isn't even a channel monitor screen written for this screen size at this point.

3djc, 2017-02-05:
@mythreeleggedcat please be also aware that the pc simulator does NOT take into account your firmware options, it is a running a default configuration

mythreeleggedcat, 2017-02-05:
Kilrah - I think you meant there is no Mixer Monitor screen for the Q-X7 in your first comment.  Which is fine if that is a limitation...but you can build the firmware for the Q-X7 with that option which is pointless if it is not there...so it is misleading...

There is definitely a Channel Monitor and in the Q-X7 Simulator and I assume on the Q-X7 and the Model editor shortcut Menu key does not work in the Simulator.  If this is the case on the radio I would say it is more important bug than implementing the Mixer Monitor.

3djc - I thought the Mixer Monitor not working on the X9D+ Simulator in 2.2 might be that Companion is not using the build options but it works in the 2.1 simulator for the X9D+....perhaps 2.1 assumes it is a configured feature even if it has not been configured and 2.2 doesn't?  Seems an unnecessary change if so.

Is there an enhancement to make the Simulator use the current firmware settings?  I think that would be useful.

3djc, 2017-02-05:
The mixermon option for X7 has been removed in next rc

kilrah, 2017-02-05:
The outputs are displayed in the lower half of some main views on the X7, but there isn't a dedicated channel monitor screen that can be called from wherever (and serve as mixer monitor) like on X9D for that screen layout, that would have to be written.

We're still in RC and the couple of simulators that are supplied for testing (there isn't even one for every radio type) just have the default options configuration selected.

Simulator that uses your options needs infrastructure to be put in place so that they can be build and downloaded on demand like the frmware. That's the plan but with no ETA.

mythreeleggedcat, 2017-02-06:
Thanks for the responses...I see now the Q-X7 Channel monitor is combined with a few other fields so is not a dedicated display... If the absence of the shortcut  is part of the same limitation which this "enhancement" addresses, then fair enough.

schubi74, 2019-08-11:
Hello,
I own an X-Lite, the second radio with the X7 resolution of 128x64. Are you aware that there is another radio on the market without a mixer monitor? I really miss it a lot when programming on the field.
Are there any actions for implementing this feature?
Regards
Andreas

closed 2020-05-07 by stale[bot]