opentx/opentx #4403: Companion Simulator issues

projectkk2glider, 2017-02-05:
Couple of issues with the new simulator at fcadd48444abe3d6e9fc2b789903f1d61ca5dab4 on Linux Mint 17.3 (basically an Ubuntu 14.04) with Qt 5.2.1

1. Different (white) background used for special texts:
 
[image]

2. Every time I focus (click) on `debug log` window I get bunch (hundreds) of these in console (CMAKE_BUILD_TYPE:STRING=Debug):
```
QPainter::setBrush: Painter not active
QPainter::drawPath: Painter not active
QPainter::setBrush: Painter not active
QPainter::setOpacity: Painter not active
QPainter::restore: Unbalanced save/restore
QPainter::end: Painter not active, aborted
QPainter::begin: Paint device returned engine == 0, type: 3
QPainter::save: Painter not active
QPainter::translate: Painter not active
QPainter::scale: Painter not active
QPainter::setPen: Painter not active
QPainter::setBrush: Painter not active
QPainter::setRenderHint: Painter must be active to set rendering hints
QPainter::setRenderHint: Painter must be active to set rendering hints
QPainter::opacity: Painter not active
QPainter::brush: Painter not active
QPainter::pen: Painter not active
QPainter::setPen: Painter not active
QPainter::setOpacity: Painter not active
```

3. I cant seem to get the filter to work, whateve I put in it (even prepared filters) I get all output:

[image]

I will try with Qt5.7 now...

kilrah, 2017-02-05:
> I cant seem to get the filter to work, whateve I put in it (even prepared filters) I get all output:

I believe the "filter" button acts a as an enable/disable

projectkk2glider, 2017-02-05:
You are right!

projectkk2glider, 2017-02-05:
Another issue:

3. Outputs window behind main Companion window:
 * open Simulator and open Outputs window (undocked)
 * close Simulator (the state of Outputs window is remembered)
 * open Simulator, Outputs window open, but is behind all windows (behind main Companion window)
 * clicking on Outputs brings it on top, but as soon as you click in the Simulator it goes behind the main Companion window again. 

[image]

projectkk2glider, 2017-02-05:
On Qt5.7 I get **all issues except problem no 2.** but the issue 1. does not look so bad.

[image]

mpaperno, 2017-02-05:
Thanks for testing Damjan.  Should I add some better indicator about the Filter label also being the en/disable button?  Or a separate control for that?  Was just trying to save horizontal space there, really.

The help text background I think is the same in both versions, but the default background of that dialog changes so it is less contrasting in the latter (the bg color is system default or whatever Qt uses).  It's just some CSS controlling it... perhaps I can find a standard palette color which would be more consistent (but not sure it's worth the hassle).

`<style>kbd {background-color: ghostwhite; font-size: large; white-space: nowrap;}</style>`

I do have Qt 5.5 here so I check out those warnings you're seeing in Debug mode.  I'm not doing any direct painting in the debug output though, so this may be a Qt bug (I didn't see it in a 5.5 non-debug build).

Finally the window position when launched from Companion could be an issue...  Not sure I've ever actually used a MainWindow as a modal dialog before.  Seems to work otherwise, but I didn't notice the problem you mention.   Will see what can be done.

mpaperno, 2017-02-05:
I just realized you said Qt 5.**2** earlier, not sure where I got 5.5 from.  5.2 is ancient news!  :)

BTW all my Linux testing is done on Mint 17.3 also.  But with downloaded Qt packages, not native.

mpaperno, 2017-02-05:
> 3. Outputs window behind main Companion window:

Apparently a "Linux special" issue and not only affects launching from Companion but from terminal also.  If any of the "remembered" floating dock widget positions are behind the terminal window, they also get hidden back there (though in that case once you activate them once they do stay on top).

Anyway, workaround found, new PR soon for this and some other niggles that have come up.

mpaperno, 2017-02-06:
Damjan, could you verify the fixes when you get a chance, esp. #3?  Then we can close this one.  Thanks!

projectkk2glider, 2017-02-12:
Issue 3 is OK now.

projectkk2glider, 2017-02-12:
Issue 1 also looks nicer.

closed 2017-02-12 by projectkk2glider