opentx/opentx #4404: Horus RC10 362 No Telemetry working with D8R II Plus

Githubbachelor, 2017-02-05:
I installed RC10 362 for Horus and Telemetry does not work with D8R II Plus A2. There is no chance to activate already existing models. And also not possible to find Sensors. After I go back to NB360 and all Telemetry works.

projectkk2glider, 2017-02-06:
I find this hard to believe, we would have a lot of similar reports if the problem was in OpenTX.

projectkk2glider, 2017-02-10:
I just tested this with current `next` on my 9XD and D8R II - works without any problems.

kilrah, 2017-02-10:
Fine on Horus here too

Githubbachelor, 2017-02-10:
Hallo Kilrah. Ok. I installed again 362 to Horus yesterday. I did this with another person with horus parallel at phone together with same results. After switching on, searching for my Model Helicopter ADAC and connecting the heli to battery it does not find my telemetrie and does not show RSSI. But I can fly the heli. After I switch off and on again and my heli shows telemetry. After I search for a multi modul model KN WLTOYS protocol and bind that. Than I choose again my heli model and connect the heli to battery and again I have no telemetry. Than I switch off the Horus and on again and the Heli shows Telemetry. Than I take another Heli with telemetry and D8R II Plus with telemetry on A2 and it does have also no telemetry. After I switch off Horus and on again and telemetry works????? That problem I do not have in 360 RC8

kilrah, 2017-02-10:
What module are you using to connect to your D8 receiver, the internal one or the multi? Please be clear and precise. Make sure your multimodule is OFF when using the internal module. Try power cycling when you swap between a model using multi and internal.

Githubbachelor, 2017-02-10:
The internal for my D8R II Plus. When I use my Multi Modul 4in1 and go back to other Models with internal Modul. I must switch of the TX after choosing a model and switch on again to get telemetry.

kilrah, 2017-02-10:
Then this matches some reports of the other direction having issues.

commit 3cd9c293 (2017-02-13): Fixes #4404
commit 777707db (2017-02-14): Fix telemetry protocol switching on Horus (Fix #4404) (#4450)

* Cosmetics

* Fixes #4404

* [9XR-PRO] Telemetry bug

* DMA apparently doesn't like being reinitialized before a deinit
closed 2017-02-14 by kilrah