opentx/opentx #4407: Companion, Does not accept any settings internal HF module from the Horus

raso12, 2017-02-05:
Companion 2.20N362
Does not accept any settings internal HF module from the Horus
No data exchange values in the data fields
The data of the models were taken over for the first time by Horus

Raimund

3djc, 2017-02-06:
Could you give us more details, we do not understand your issue

raso12, 2017-02-06:
I have for the first time models from the TX Horus
loadet in companion Companion 2.20N362
(Reading models and settings from TX)

When then I open a model in companion (witch reading before), then the
Fields for internelly HF-modul The field does not contain any values

In TX Horus I entered internally mode D16 with CH1-CH8
But the values are not passed to Companion  2.20N362
(Reading models and settings from TX)

I hope you understand.

thank you 

Raimund

raso12, 2017-02-06:
the Value from the field internally HF-Modul is

internally Protokoll "off"   this is a bug

Tbolt65, 2017-02-07:
I noticed the same issue.  Companion would turn Off the Protocol for the Internal Radio System.   I closed the issue because on a later attempt I could duplicate the problem.  20+ models loaded; not all had the change made by Companion; and it seemed to affect D16 more so then D8 settings (although that doesn't make sense to me).

projectkk2glider, 2017-02-19:
I can't reproduce. @raso12 or @Tbolt65 if you manage to reproduce this, then send us a copy of your radio/models, this is how:
 * connect powered on Horus to the computer
 * open Horus drive in file manager
 * add folders `RADIO` and `MODELS` to a zip archive (create new zip archive) and upload that zip file somewhere (like http://www.filedropper.com/) and post link here.

Do not send `.otx` file - because that one will already have the settings changed.

raso12, 2017-02-19:
Hi

the link from raso12

http://www.filedropper.com/horus

Raimund

-- 
Raimund Sonst
Nurflügel & Modelltechnik
Ravensberger Str.15
32107 Bad Salzuflen
rsonst@t-online.de
Tel.: 05221/759330
www.nurfluegel-modelltechnik.de

=============================================================

Diese eMail enthaelt vertrauliche und/oder rechtlich geschuetzte Informationen. Wenn Sie nicht der richtige Adressat sind oder diese eMail irrtuemlich erhalten haben, informieren Sie bitte sofort den Absender und vernichten Sie diese Mail. Das unerlaubte Kopieren sowie die unbefugte Weitergabe dieser Mail ist nicht gestattet.
This email may contain confidential and/or privileged information. If you are not the intended recipient (or have received this email in error) please notify the sender immediately and destroy this mail. Any unauthorized

projectkk2glider, 2017-02-19:
Reproduced!

Loading provided file via `Read models and settings`:
 * default model is `Gotha` which is in `model10.bin`
 * the internal RF in Companion is `OFF`
 * on the radio: `D16` `CH1-8` `Hold` `Internal`

I've created a smiplified setup with only one model http://wikisend.com/download/770598/issue_4407.otx Loading this in Companion produces the same result.

bsongis, 2017-03-02:
@projectkk2glider 404 not found for your otx simplified file :(
@kilrah do you have it?

projectkk2glider, 2017-03-02:
New upload http://wikisend.com/download/934984/issue_4407.otx

bsongis, 2017-03-02:
Yes, interesting!

commit 6f0e5835 (2017-03-04): [Horus] Fixes #4407 (#4543)

* [Horus] Fixes #4407

* Tab replace by spaces
closed 2017-03-04 by bsongis