opentx/opentx #4415: RC10 N362: It does not accept any file with more than 6 letters in the name.

tubeme, 2017-02-06:
I put my custom WAV files. If larger than 6 letter name that it does not appear in the track list.

With the JPG's if the file is with larger than 6 letters in the name, it is visible you can select it as model icon but then it does not load. Once I renamed the files to 6 letter names then no problem. Sounds load, jpgs are visible.

kilrah, 2017-02-06:
The 6-letter limit is normal.

>With the JPG's if the file is with larger than 6 letters in the name, it is visible you can select it as model icon 

Where? In companion? On the radio? in the model setup page or in the SD manager?

dk7xe, 2017-02-08:
I guess in companion at the model image file selection he means.
[image]
But in fact if the file name is longer than 6 char then the extension is cut off. 
in my Example "monster.jpg" was becoming "monster.jp"

tubeme, 2017-02-08:
The JPG files are visible in the radio in the model setup page drop down. But then once selected nothing appears on the model screen.

3djc, 2017-02-09:
There is at least an issue on Horus firmware:
[image]

PR #4432 (2017-02-10): Horus : limit model bitmaps filemane to 6 chars
Limit filename to 6 as sdcardfilelist controls size without extension (and size with ext was sent).

This closes #4415
closed 2017-02-10 by kilrah