opentx/opentx #4416: RC10 N362 on Taranis Q X7: not possible to set a logical switch using telemetry value 'cells' as V1

JR63, 2017-02-06:
First of all: Many thanks for openTX on the brand new Taranis Q X7!

I found that it is not possible to set the telemetry value 'cells' (and maybe others) as V1 for a logical switch.

'Tmp1+' is the last telemetry value that could be choosen, after that the entry wraps around to '---'

[image]

3djc, 2017-02-06:
Have you discovered Cells sensor, because I think it does work.

JR63, 2017-02-06:
wow, fast response, thanks.

Yes I discovered it:

[image]

3djc, 2017-02-06:
Then I'll need your model file, I cannot reproduce it here

kilrah, 2017-02-06:
Meh ;)

closed 2017-02-06 by kilrah

JR63, 2017-02-06:
Many thanks, so my model file is not needed anymore?

3djc, 2017-02-06:
Nop, next RC will fix it

JR63, 2017-02-06:
Cool, many thanks, good work!

commit 1e0046f9 (2017-02-06): Leftover cast (Fix #4416)