opentx/opentx #4419: RC10 N362: Simulator switch and slider layout (TARANIS & HORUS)

mizuta1951, 2017-02-07:
The switch layout on simulator dose not match the real Tx layout.
In simulator switches from left to right are SA, SB, SC, SD, SE, SF, SG, SH.
But on real Tx it is SF, SE, SA, SB, SC, SD, SG, SH. as it was on V2.1 simulator.
Also the sliders on HORUS simulator is L1, L2, LS, RS.
Should be better like LS, L1, L2, RS.
And on TARANIS simulator the sliders are gone.

mpaperno, 2017-02-07:
Please see https://github.com/opentx/opentx/pull/4385#issuecomment-277315610 and response https://github.com/opentx/opentx/pull/4385#issuecomment-277318593

Can you post a screenshot of Taranis simulator w/out sliders?  Also (if they're really missing) do you have sliders configured in the radio hardware settings (eg. as per default)?

mizuta1951, 2017-02-07:
Confirmed hardware settings on Taranis, is was set as "NONE".
Slider appeared after setting to "Slider with detent" , it was my fault.
It is not a bug, just a little inconvenience.
The importance is not high, I entrust it to future discussion

closed 2019-11-15 by stale[bot]