opentx/opentx #4426: RC10 (N362) Companion(HORUS)　:　SH strange behavior on logic switch

mizuta1951, 2017-02-08:
Only on Companion, dose not happen on HORUS.
Example : L1　SA↑ AND SH↓
If SA is ↑  and SH is ↑, when you start the simulator L1 is OFF (correct).
Once you move SH to ↓, L! is still OFF (wrong).
Then release SH to ↑, L1 becomes to ON (wrong).
After this L1 behave as SH is reversed.

mpaperno, 2017-02-08:
I'm pretty sure this is fixed by cab26a6 .  But just to confirm, do I have it set up right?  This is after several restarts and multiple SH toggles, still shows correct.

[image]

mizuta1951, 2017-02-08:
Strange,,,
Different behavior.
I'm using Win-10, is that　cause something ?

[image]
[image]

projectkk2glider, 2017-02-08:
The inverted SH problem is fixed by #4351

mpaperno, 2017-02-08:
Ah right, I knew it was fixed somewhere along the line.  Anyway, confirmed works in latest version with same setup as @mizuta1951, so fix will be in next RC.

closed 2017-02-08 by mpaperno