opentx/opentx #4431: Special Funcion: Set GPS Home Position (wish)

Allerhopp, 2017-02-09:
Parameter: (Which) GPS

Useful in following situations:
- can be triggered right before start to have best possible accuracy
- better handling and more targeted than "Telemetry reset"
- GPS (e.g. openXsensor) can transmit number of sat and/or HDOP to trigger only a accurate GPS HP
- TX onboard (SPort) GPS can update GPS HP timer triggered (e.g. for cross country)

1Sven, 2017-02-09:
Oh, this is a perfect idea! I hope we are heard...

projectkk2glider, 2017-02-10:
Nice ideas, it will require an EEROM format change. Other possibility would be a new Lua function.

>TX onboard (SPort) GPS can update GPS HP timer triggered (e.g. for cross country)

Could you explain this, not sure what you mean.

Allerhopp, 2017-02-10:
http://openrcforums.com/forum/viewtopic.php?f=86&t=9412
It is a way make the Horus TX sensors GPS and accelerometer available for X7 and X9 radios. It is already working, the only problem is the Home- or Pilot-Position update. If you want to know the real distance to the plane, moving yourself with the TX (cross country flying for example). A LUA function can do the job too. In my amateurish thinking it only needs a (over)-write of the Pilot-Coordinates in the telemetry data array or database or whatever you pros call it.

3djc, 2017-02-10:
Are you running while flying or something like that, because I still cannot understand the need, sorry

Allerhopp, 2017-02-10:
Yes, you walk or drive and follow your glider. The record is near 300 km.

3djc, 2017-02-10:
Then I think you probably want a radio with GPS inside rather than this solution, as it seems nearly impossible to assess when a flying device is straight above you to get a proper home position (which would always move if you are in a driving car anyway

Allerhopp, 2017-02-10:
There is a GPS inside the radio, connected to the SPort in the module bay. It needs a different sensor ID to work, but it is tested already. Both GPS can be logged, the only disadvantage is that the pilot position can not yet be updated. 
https://www.youtube.com/watch?v=p-cMhuEyLfk

3djc, 2017-02-10:
Ok, so what you need is the capability to set an "external" GPS to be treated as internal really, no ?

kilrah, 2017-02-10:
If the home position was associated with the telemetry sensor, it would only be needed to reset said sensor. No EEPROM change needed.

Allerhopp, 2017-02-10:
There is a GPS in the Glider "GPSair" and a GPS in the TX "GPStx". I can see both in the telemtry screens, because they have different sensor IDs. I can log both GPS. If i move with the TX, it would be great to have a chance to update the pilote position with the coordinates of "GPStx". GPS distance would calculate always the true distance between "GPSair" and "GPStx".

Allerhopp, 2017-02-10:
Never thought about this, but it should work too!

1Sven, 2017-02-10:
Is it not possibele to make GPS as sensor and the inbuild mems as input? (Horus)

kilrah, 2017-02-10:
> If i move with the TX, it would be great to have a chance to update the pilote position with the coordinates of "GPStx". GPS distance would calculate always the true distance between plane ="GPSair" and pilot ="GPStx".

The internal GPS is already available to Lua for people to be able to write exactly that kind of scripts.

Allerhopp, 2017-02-10:
Not sure, if you mean Horus or Taranis. But no matter, in both cases you are right. The disadvantage is, that the LUA calculated distance does not appear in the log. Is there a chance, to overwrite pilote position from LUA? Or make it accessible for LUA?

kilrah, 2017-02-10:
2.2 lets you define custom sensors from lua that can be logged along the others. You can also write your own logs from lua since you can write to files.

Allerhopp, 2017-02-10:
Great news, especially the first part, because there is no need to merge the logs afterwards. Can you point me to a description or are there comments in the code?

kilrah, 2017-02-10:
Refer to the doc... https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/content/general/setTelemetryValue.html

Allerhopp, 2017-02-10:
My LUA Ref bookmark missed the autoupdate ;)
Thank you!

closed 2017-02-10 by Allerhopp