opentx/opentx #4434: Missing sensor in telemetry simulator companion

1Sven, 2017-02-09:
The simulator is really fine. But when make discover, only sensors will seen as in simulator window available.
Like the RB there is no value visible. But the log-file is fine.
[image]

[Model12-2017-02-09.zip](https://github.com/opentx/opentx/files/764737/Model12-2017-02-09.zip)

mpaperno, 2017-02-09:
Well the telemetry simulator could certainly use some attention in terms of specifying new/custom sensors and in a few other areas.  But can't you just add the missing sensor(s) manually to the radio config for now?  Or am I not understanding the problem?

kilrah, 2017-02-09:
I think he might want to replay a log which contains values for some sensors that aren't supported by the telemetry simulator...

mpaperno, 2017-02-09:
Ah, yes, I see that all the possible sensors/headers are also hard-coded for parsing the CSV log.  And RB is not in the list.  We could keep adding "known" sensors to the list and UI, but really a lot of the telemetry simulator needs to be re-written to be more dynamic and user-configurable.

"RB" = Redundancy Box?

1Sven, 2017-02-10:
Yes, it's the Redundancy Box. The T-Sim is the best way for testing new widges. In this case I will use the ECU-Converter, made by Raphael. The sensor-values must chage to status and so on. For testing in the script it's a perfect way with thw T-Sim. (...so don't need running the turbine)  I think it will be good, when the T-Sim automaticly performs the sensors there be used in the CSV log. Or, a window for create a sensor by  ID...

1Sven, 2019-08-17:
Please do not leave this. May be it is possibele to make a "hole" for "private" sensors...

closed 2020-05-13 by stale[bot]