opentx/opentx #4438: Mixer script source selector only has a few telemetry values

RealTadango, 2017-02-11:
Mixer script source selector only has a few telemetry values. I only see 3 while i have about 20 sensors....

kilrah, 2017-02-11:
Companion? Radio? What radio type? What version?

Please....

RealTadango, 2017-02-11:
Horus, on the radio. Code from 03-02-2017

3djc, 2017-02-11:
Having a list of what values are available, and which not could also help debuging

RealTadango, 2017-02-11:
I saw RSSI, RxBT, Tmp1
I missed RPM, ASpd, RB1/2 V/C, other custom sensors

Will test with latest next but did not see any changes related.

RealTadango, 2017-02-11:
Same result with latest NEXT built

projectkk2glider, 2017-02-16:
This is in fact not a bug. Any telemetry source can not be selected as a source for a mixer. You have to go trough inputs where you can use telemetry value as a source. This is how the system is supposed to work. Not sure how you saw RSSI, RxBT, Tmp1, none of them should be present in mixer source selection.

In any case I just verified Taranis and Hours in current (ee43d3fb971aef5295ea70a50dac9e94b74b0bdf) Companion and I see no problems.

projectkk2glider, 2017-02-16:
Reopening, I misread, the issue is about `Lua mixer scripts`.

projectkk2glider, 2017-02-16:
Confirmed in Companion Horus simlator

RealTadango, 2017-02-16:
I just noticed it jumps to --- after the 3th sensor but with only rxbatt and rssi it remains at the last (as it should) so the bug is not missing sensors in the selection but an error in the selector itself now allowing to select a sensor....

kilrah, 2017-02-16:
Just wrong type since there are now >256 possibilities in the source list. Being fixed.

RealTadango, 2017-02-16:
Solution works fine :)

PR #4470 (2017-02-21): Fixes #4438: wrong cast used when setting Lua mixer script input source
Improved parsing of Lua mixer script inputs

Closes #4438
commit f36d01d5 (2017-02-21): Fixes #4438: wrong cast used when setting Lua mixer script input source (#4470)

* Fixes #4438: wrong cast used when setting Lua mixer script input source
Improved parsing of Lua mixer script inputs

* Additional type check

* ScriptDataInput union introduced.
Fixed: when using SOURCE input, do not use default value
Fixed: when using SOURCE input, min-max range is fixed
closed 2017-02-21 by bsongis