opentx/opentx #4439: OpenTX 2.2RC 10 won't save timer settings

Dutchy-79, 2017-02-11:
Hi,

I tried setting 3 timers on my model but it won't save the settings,
Added some pics in this gallery, first shows after first boot then 2 setup timer, timer is there, reboot timer gone again.

Ps, BMP not supported while Taranis screenshots come as BMP so made a online gallery

https://postimg.org/gallery/27pf4vvgq/
[S500.zip](https://github.com/opentx/opentx/files/768778/S500.zip)

Regards,

Jasper

Dutchy-79, 2017-02-11:
Edit, could have made a zip as well
[SCREENSHOTS.zip](https://github.com/opentx/opentx/files/768780/SCREENSHOTS.zip)

projectkk2glider, 2017-02-11:
Can't reproduce. Are you doing this on radio or is Companion involved? Which setting is not saved and for which timer?

Dutchy-79, 2017-02-11:
Timer 1 is not saved, using the radio menu and the model in the zip file.

Dutchy-79, 2017-02-11:
I created the model while I was using 2.1.9 but don't know if this is the reason. Many others probably had the same problem if that's the case.

projectkk2glider, 2017-02-11:
And which parameter is not saved?

Dutchy-79, 2017-02-11:
Timer 1 is now off and I want it to be THs 
It works as long as I don't turn off the radio. 
When turning back on the timer is OFF again, tho it does keep the name saved. 

Can it be caused by a outdated Lua script? I do use a teensy for apm telemetry which uses a Lua script.

projectkk2glider, 2017-02-11:
Yes, Lua script could modify the timer.

Dutchy-79, 2017-02-11:
Just noticed that also minute call turns on everytime for timer 1.

Ill drop all Lua tomorrow and then report back.

mpaperno, 2017-02-11:
@Dutchy-79 could you try this on a fresh model?

Dutchy-79, 2017-02-11:
Just deleted all Lua scripts and it works, so luckily no bug just another user error, replace user,...

closed 2017-02-11 by Dutchy-79