opentx/opentx #4443: Companion 2.1.9 Model Edit Setup Failsafe Position Rounding

elecpower, 2017-02-11:
Most channel failsafe position values rounded down 0.1 when saved after changing.
Open and close edit dialog without changing a failsafe value preserves the values as read.
Only rounds the channel being changed. Changing from a starting value and then back to the same value triggers the rounding issue,

eg 20.0 saved as 19.9 and 20.1 saved as 20.0
but 10.0 saved as 10.0 and 10.1 saved as 10.1

Taranis Plus model with Internal D16 Protocol
Companion 2.1.9
Tested on Win10 and Ubuntu 14.04

powercycle, 2017-02-12:
Confirmed on Companion 2.2 RC10, Ubuntu 16.04.

powercycle, 2017-02-12:
Does this happen on a MAC?
onFailsafeSpinChanged seems to be storing the wrong value.

mpaperno, 2017-02-12:
Yes, this is a result of rounding.  The actual failsafe value is stored as an integer in PWM μs.  The underlying issue is that the edited value being displayed doesn't actually reflect the final value which will be stored (due to the rounding).  Any further edits after saving will compound the problem.  

There is an even deeper issue in that with Companion you can set a value which is out of range on the actual radio.  The min/max value of that spinbox in Companion doesn't change with the extended limits setting right above it.

I'm not even sure what the value in the spinbox is supposed to represent.  Percent at extended limits?  The allowed range is -150 to +150.  Setting it to maximum gives a 2268μs pulse width (which does correspond to absolute maximum with extd. limits on).

Also when editing custom failsafes in the firmware (radio/simu), the min/max with extended limits doesn't match Companion.

Firmware: `int lim = g_model.extendedLimits ? 640*2 : 512*2;`

Whereas in Companion min/max with ext. limits is (-/+) 768.0us.

I've got it all open and am working on it but not yet sure of the best approach to reconcile all these disparities.

3djc, 2017-02-12:
And even further, how do we ensure they KEEP consistent

elecpower, 2017-02-12:
Is there a possibility with the current settings that the failsafe positions can be set outside the physical limits of the servos for the model in question (not servo's manufacture limit) thus over driving them? This could lead to even more problems at the time of failure.
Maybe in any rework of this issue, the channel output min/max and extended limits be factored into the calculation of the final pulse value. Possibly the spinbox is always a percentage -/+100 of the permissible range and the value sent to the receiver is recalculated every time there is a change to any of the effecting parameters. A -/+100 range would keep it nice and simple for the pilot and they should not need a maths degree to calc the failsafe position.

kilrah, 2017-02-12:
Of course. 
It makes little sense adjsuting the failsafes in companion anyway, it's useful to have them to see whether you've set them more or less reasonably but you woulnd't use that for actual adjustment apart maybe for simple things like electric motor throttle.
You should be using the facilities on the radio that allow for real-time adjustment of positions with the actual controls moving where you want them to, and of course check failsafe operates properly.

mpaperno, 2017-02-12:
I'm adjusting it so the Companion UI follows the ext. limits setting and (I think) can show PPM μs instead of % if that firmware option is selected (personally PPM_US is one of the first options I set, avoids any ambiguity with %s in my mind),  Also fixing the display so that it shows correctly rounded values (WYSIWYG).  I think it's reasonable to expect the setup UI to at least be correct in those regards.

Beyond that, I think Andre is totally right, setting up failsafe should be done with actual response from the model.  If you (the pilot) change any relevant settings, you're in the best position to know when to re-do the failsafes.

PR #4492 (2017-02-26): [Companion] Fix/improve setting radio module custom failsafe values.
  * Properly set limits based on model's Extended Limits setting;
  * Fix rounding issues between percentile display and underlying value in μs;
  * Add ability to display values in μs or percent.
  * Channels numbering & count is responsive to module's defined start/end channels.

Fixes #4443

Please give this a good test first... I'm a bit rushed on final testing, and may be not be available for a few days for quick fixes.  I did give it a good workout, but you never know.  Thanks!
commit 171e08c2 (2017-02-26): [Companion][ModelEdit] Fix/improve module custom failsafe value settings (fixes #4443): (#4492)

* Properly set limits based on model's Extended Limits setting;
    * Fix rounding issues between percentile display and underlying value in μs;
    * Add ability to display values in μs as well as percent.
    * Channels numbering & count is responsive to module's defined start/end channels.
closed 2017-02-26 by bsongis