opentx/opentx #4444: Lua getModule() function should also return module mode.

mrpraline, 2017-02-12:
Currently the getModule() function returns only the RF protocol, not the RF mode. The value that is currently returned for rfProtocol is 0, regardless if the RF mode is OFF or any other value. In companion getModule(x).rfProtocol always returns 0,  I don't know what happens in a real transmitter without an external module.

As a consequence, it is currently impossible to determine if a module is actually transmitting and thus what channels  are physically being transmitted to the model. Adding the mode to the table result table would solve this. 

Background:
I want to create a one time script to reorder the physical outputs on the receiver. For that only the channels that are actually being transmitted are of interest. The user should not be able to choose other channels and also limiting the amount of channels will greatly save screen space in most circumstances.

0815Creeper, 2019-11-11:
The RF protocol has apparently been renamed to subType but still does not match the documentation of model.get/setModule(): It reads 1 for D8, 2 for L12 and 0 for anything else (including OFF, which should be -1 according to doc...)

kilrah, 2019-11-11:
There are now so many possible combinations and values for module settings it's almost impossible to maintain such a function at all.

closed 2020-08-07 by stale[bot]