opentx/opentx #4466: No discovered sensors for 2.1 on XSR

Harlequinfpv, 2017-02-16:
Smart port is hooked up and correct, however, I do not even get a RSSI reading or FC voltage when trying to discover sensors. Running 2.1 on a Taranis Plus

3djc, 2017-02-16:
Please do try with just the receiver alone powered and connected to nothing but power supply (specialty sport)

Harlequinfpv, 2017-02-16:
Tried just powering RX and still no discovered sensors

3djc, 2017-02-16:
Not sure one of us as an XSR to test, but since telem is not receiver specific, I cannot think of a reason why this receiver would be an issue, unless your XSR is having an issue

Harlequinfpv, 2017-02-16:
Tried 2 XSRs and both don't seem to show the standard RX sensors or Telemetry though smart port on the Uartv . Basically can't get any telemetry to work at this moment.

Harlequinfpv, 2017-02-16:
Finally found a thread with the same issues.... looks like I may have a hardware issue inside my taranis.

schwabe, 2017-02-16:
My XSR is working just fine.

closed 2017-02-16 by bsongis