opentx/opentx #4473: LUA Script syntax error on RC10 (works fine on RC9)

slawek89, 2017-02-16:
It is some kind of bug that Betaflight LUA script dont work on OTX 2.2 RC10? Everytime show syntax error.
On RC9 everything works fine.

1Sven, 2017-02-16:
The script works on Horus. I had today tested.

kilrah, 2017-02-16:
What radio?...

slawek89, 2017-02-17:
Sorry, it is for taranis.

mpaperno, 2017-03-03:
More details?  Or can this be closed?  

Personally I have no idea what "that Betaflight LUA script" entails or why anyone would assume otherwise.

kilrah, 2017-03-03:
Especially since there are several variants of the script floating around.

slawek89, 2017-03-24:
Sorry for long time reply. This issue can be closed.
It turns out that Betaflight LUA script was divided to two files and they has to be in proper localisation.

closed 2017-03-24 by slawek89

3djc, 2017-03-24:
Thanks for the update !