opentx/opentx #4479: USB joystick lag since upgrading to 2.2 firmware on Taranis X9Dplus

Enlightx, 2017-02-18:
Iv been using a lot of emulators and had no issues all input was smooth.
Since updating to 2.2 firmware iv found that USB joystick input is not smooth.
when moving throttle quickly you can hear input it sharp stepped and not smooth.
Smooth if you throttle slowley but sharp movements as its as its skipped steps and throttle is not smooth because of not getting the data quick enough to the pc?

is it a known issue or can someone test see if they have the same?

projectkk2glider, 2017-02-19:
I would like to have a confirmation from another user before going into this.

Mistale, 2017-02-23:
I'm having similar issues. Ive been using my X9d with Velocidrone simulator without a hitch, but after updating to 2.2, input seems to be stepped / unfiltered resulting in untunable copter in the sim.

projectkk2glider, 2017-03-06:
I don't see any problem on Linux, please provide an example of the good and bad performance. Here is an example how it works on my system: https://www.youtube.com/watch?v=rNaxqF5JQUY

Enlightx, 2017-03-06:
Is it possible to downgrade the flash? Then I could try on 2.1 and 2.2

projectkk2glider, 2017-03-06:
Only if you stored both the 2.1 firmware and radio/models settings. When flashing back 2.1 your 2.2 models will get lost since OpenTX only supports model conversion when going from old to newer version and not when going in opposite direction.

projectkk2glider, 2017-03-06:
This might be Windows specific problem and that might explain why I don't see the problem.

Enlightx, 2017-03-06:
Let me get it tested on a few other computers and older windows version

Problem seems to be rapid movements of sticks created lag. Think I might have a screen recording before I upgraded to 2.2 I'll have a look

projectkk2glider, 2017-03-06:
Do you see the same lag in Windows Joystick calibration app?

Enlightx, 2017-03-06:
Yeah same in calibration app . Tried recalibration but same effect also

Windows 10 64 bit

kilrah, 2017-03-06:
Can't seem to see an issue here on Win... feels no slower than another controller.

projectkk2glider, 2017-03-06:
The only issue I see so far here is that actual USB joystick packets get send out every 50 - 60ms. Investigating why...

projectkk2glider, 2017-03-06:
Test firmwares (English language, USB Joystick, 3 Timers, Lua):
 * Taranis http://downloads.open-tx.org/2.2//tests/opentx-x9d-lua-double-fix.bin
 * Taranis Plus http://downloads.open-tx.org/2.2//tests/opentx-x9dplus-lua-double-fix.bin

projectkk2glider, 2017-03-06:
Test firmwares (USB packets sent every  10ms):
 * Taranis http://downloads.open-tx.org/2.2/tests/opentx-x9d-joystick-fix.bin
 * Taranis Plus http://downloads.open-tx.org/2.2/tests/opentx-x9dplus-joystick-fix.bin

@Enlightx Try one of these firmwares (depending which Taranis you own) and tell me if it is any better.

Enlightx, 2017-03-07:
Hi
Tested on the X9 Plus

Completely fixed! thank you

Throttle is now super smooth when rapid movements occur.

I did take videos to show difference if they are any use to you.

Thanks again
Luke

projectkk2glider, 2017-03-07:
Thanks for the report, videos are not needed.

PR #4569 (2017-03-07): Projectkk2glider/issue 4479 slow joystick
~~Wait for feedback in https://github.com/opentx/opentx/issues/4479#issuecomment-284531024 before merging.~~

Closes #4479
closed 2017-03-07 by bsongis