opentx/opentx #4480: T5 / T6 Horus => Companion transfers that sources as "Lua1a"

Casichispea, 2017-02-19:
if you have created a model in the Horus using  T5 as source, when you copy it with companion 2.2 into the computer, the process changes the source "T5" and appears  "LUA1a".

Red-Devil74, 2017-02-22:
As T5 and T6 are not present in Companion, I assume it defaults to the first entry in the list "LUA1a". I have the same issue. After transfer from Horus to Companion it shows 3 question marks in the mixes tab. Only after opening said mix is appears as "LUA1a". (hope this info helps a bit...)
[image]
[image]

Red-Devil74, 2017-02-27:
Is this not fixed as per remark in buglist "Horus misc issues #3393"?

kilrah, 2017-02-27:
No that referred to radio firmware only, still missing in companion

closed 2017-03-05 by kilrah