opentx/opentx #4489: Trim values have a maximum of 250 instead of the expected 256

Miami-Mike, 2017-02-20:
Trims have a maximum integer value of 250 (1/4 of 1,000) instead of the expected 256 (1/4 of 1024). Has this been corrected yet for version 2.2?

3djc, 2017-02-21:
Trims in 2.2 have the same - 125 to + 125 range,  that can be extended to - 500 to +500 by selecting extended Trims.

kilrah, 2017-02-21:
The +/-125 "tick" range does map to +/-25% travel. That's how it's always been.

Miami-Mike, 2017-02-21:
Sorry, I should have been more explicit. First, here's simple evidence of the bug:

Run the trim up to maximum on any stick value that's centered and you'll see a value of 24.4%. This displayed value is actually 250 / 1024 = 0.244140625 = 24.4140625%, and appears to be the result of a programmer confusing the percentage value of 25% with the integer value of 250. The result of this error is that a mix such as _TrmA Weight(+25%)_ produces an integer value of 256 while full trim only adds a value of 250. These results can be viewed in a Lua script that reads getValue(source).

I've found ways to work around this in the work I'm doing with trims, but it's still something that ought to be fixed.

Miami-Mike, 2017-02-21:
P.S. - Here's an oddity that fixing this bug will cure: In any trim step setting beside "Extra Fine" (2 steps per click), the values accessible when you start from zero are interleaved with the values accessible when coming back from maximum or minimum. For example, if your trim setting is "Fine" (4 steps per click) then if you start from zero you can select settings that are multiples of 4, such 4, 8, 12, 16, etc.

But if you go all the way to the current unintended limit of 250 and come back, you can only get to settings that are equal to 250 minus a multiple of four, such as 18, 14, 10, 6, etc.

Increasing the trim limit to 256, as it was almost certainly intended to be in the first place, would clean up that problem.

Miami-Mike, 2017-02-28:
It's been a week. Have any of you either verified or attempted to refute this report?

kilrah, 2017-02-28:
It's been that way since OpenTX exists, so certainly not something that will be changed in a pinch. Will need discussion about whether a change here is worth it or even desirable at all.

Miami-Mike, 2017-02-28:
I find it difficult to believe this trim oddity was deliberate as it makes no sense, but thanks for at least acknowledging that it exists. I've found that creating an input such as ``MAX(Weight 0%) TrmA`` is a viable workaround.

closed 2019-11-15 by stale[bot]