opentx/opentx #4490: can't change channel order

nathantsoi, 2017-02-21:
Just updated my Taranis Q X7 to rc9 and I can't set the channel order under radio setup... It is possible this is due to operator error...

Here's what I see: https://goo.gl/photos/uSSh7ZzybMdf5B167

projectkk2glider, 2017-02-21:
Please try with RC10 it might already be fixed.

3djc, 2017-02-21:
Why do you update to RC9 when RC10 as been out for a month or so (and does fix the issue you are reporting)

nathantsoi, 2017-02-21:
Ha. Thx guys. I'll update to RC10. Google fail :(

[image]

nathantsoi, 2017-02-21:
Ok, just double checked and this was actually with, 2.2.0N362

(md5) efef1be8600f65cfff6b6cefc51b8c33  opentx-x7-lua-sqt5font-en.bin

so I think this could be a real bug?

projectkk2glider, 2017-02-21:
Just checked in `simu`, appears fixed, @nathantsoi fix will be in RC11:
[image]

closed 2017-02-21 by projectkk2glider

nathantsoi, 2017-02-21:
awesome. thx @projectkk2glider!