opentx/opentx #4493: gcc 6.2.1 - logs.cpp - error: call of overloaded div

griebd, 2017-02-22:
Hi, after been away for a long time...

Got the following error trying to use gcc 6.2.1 to compile the firmware (on linux and using the master branch):
```
logs.cpp: In function 'void writeLogs()':
logs.cpp:312:52: error: call of overloaded 'div(int32_t&, int)' is ambiguous
             div_t qr = div(telemetryItem.value, 100);
```

Make options are:
`make PCB=TARANIS PCBREV=REVPLUS HELI=YES GVARS=YES MIXERS_MONITOR=YES TEMPLATES=NO`

My main question is: is gcc 6 already supported or am I on my own?

Thanks as always,

Grieb.

projectkk2glider, 2017-02-22:
It is not a question of support, the new versions of gcc are usually more sensitive to such things and give warnings that are not in previous version.  I think we fixed this in the `next` branch. Can you test that.

schwabe, 2017-02-22:
yes, thee issues are fixed in next, I have tested both firmware and companion with gcc 6 in next

griebd, 2017-02-22:
ok then, will check it! Thanks!

griebd, 2017-02-23:
I just went for next branch... but then I'm having two issues! I just want to confirm then before opening a new issue (it maybe me doing something wrong)...

1. my models have a name sound!
- in 2.1 when I turn the radio on the welcome message is played and nothing else. When I change models the name is played. All OK!
- in 2.2 when I turn the radio on the welcome and the name are played "at the same time", actually one interrupts the other... it's not a breaking error but it's not nice either...

2. the exclamation mark icon (you know which?) is always there, I can't make it disappear, no matter what...

3djc, 2017-02-23:
are you talking simu or actual radio ? What is your arm compiler version ?

griebd, 2017-02-23:
actual radio! 
```
./arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]
```

3djc, 2017-02-23:
That may be why my.  We compile radio firmware with : gcc-arm-none-eabi-4_7-2013q3-20130916

griebd, 2017-02-23:
Item 2 maybe, I will test it later.

But I don't believe the sound problem has any relation to the gcc version used. (not saying it's impossible, but quite unlikely...)

3djc, 2017-02-23:
Yes,  I was indeed referring to your second point

kilrah, 2017-02-23:
Which method are you using to play the model name sound?

3djc, 2017-02-23:
Most likely by having a wav with the same name as the model (that's what I understood from "my models have a name sound!")

griebd, 2017-02-23:
exactly that... :wink:

kilrah, 2017-02-23:
Pretty sure that hasn't been tested in a while, so could be broken...

kilrah, 2017-02-25:
Please open a new issue for the name problem, messy to continue here...

closed 2017-02-25 by kilrah