opentx/opentx #4500: companion read model doesn´t work any more

Casichispea, 2017-02-23:
everything has been working well and suddenly 
[image]

3djc, 2017-02-23:
Well if it worked ok, it means likely that the error is not in the code. Have you been removing the SDCard ?

Casichispea, 2017-02-23:
Windows could read the SDCard w/o problems. I have restart computer, Horus, with SDCard in, out....nothing help. At the end I have written an old version of an otx file in the horus and that has worked after that companion can read the models in Horus normally again.   I have lost all the changes made directly in the field yesterday in the flyfield.... Horus has been working normally during the problem with companion.

kilrah, 2017-02-23:
Did you save the card contents before restoring the old file? The only waywe could investigate is if you supplied a copy of the problematic data.

Casichispea, 2017-02-23:
I have synchroninized the SD Card and it worked. Could you find anything in the SD images on computer?

kilrah, 2017-02-23:
Only if what you have is still in the "broken" state.

Casichispea, 2017-02-23:
yes it is

kilrah, 2017-02-23:
Then you can zip the whole content and upload to a service like www.filedropper.com, and post the link here, we'll have a look.

Casichispea, 2017-02-23:
http://www.filedropper.com/n362-kopie

Casichispea, 2017-02-23:
The error can be reproduced. It occurs when I writte some wav files directly in the SD Cart in Horus using the file explorer and I did in this way because if I writte the sound files in the SD image in the computer and this files are used by a modell, when they are synchronized the are corrupted and transformed in a file with 1K size.

kilrah, 2017-02-23:
That would suggest a card problem. Did you try another one?

Casichispea, 2017-02-23:
the same again with another card and this is the order what I do:
1. I write some wav file in the Horus SD Card which are used by some models.
2. Synchronize the SD Card with companion.
3. Error comes when I try to read the models.
4 then I write and old otx file in the Tx SD Card and everything works fine again.

and the oposite direction

1. I write a wav file in the computer SD Card images.
2. Synchronize => in Horus SD are written a file with 1K size corruted

Casichispea, 2017-02-24:
Before I had taken the new card and copied the old one in the new => same result
Now I had copy the SD Card content from Internet V2017-01-25 19:14	29M	 and started from cero. Everything seems to be Ok now.

closed 2017-02-25 by kilrah