opentx/opentx #4501: Logging the flight mode

jfrickmann, 2017-02-24:
It would be really nice if we could record the flight mode in the log file.
Please, and thank you for OpenTX - it is awesome :-)

projectkk2glider, 2017-02-24:
This can be solved with Lua mixer scritpt that reads current flight mode (`getFlightMode()`) and writes it to custom sensor (`setTelemetryValue()`) that is then logged.

jfrickmann, 2017-02-25:
Thanks - looks like it is coming in 2.2, but not in 2.1?

3djc, 2017-02-25:
You have been given a way to do it in 2.2 without needing firmware change,

powercycle, 2017-02-26:
jfrickmann,

Additionally 2.1.x seems a little dated.
What makes you need to stay with 2.1.x?

jfrickmann, 2017-02-27:
As far as I know, 2.1 is the current version?

kilrah, 2017-02-27:
Would be handled in #4166, FM could be a useful thing to add to a group indeed.

closed 2017-02-27 by kilrah