opentx/opentx #4505: OpenTX 2.2.0N362 Multi Modul protocol 32 not work

profdrdenis, 2017-02-24:
OpenTX 2.2.0N362

Multi Modul protocol 32 (GW0008) cant selected. Work only to 31

schwabe, 2017-02-25:
What remote, do you mean selecting in custom protocol? I am not sure I can follow.

profdrdenis, 2017-02-25:
Hello,
the protocoll 32 (GW0008) is new maybe 1-2 days young. The complete list you see here:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multi.txt
have a look than you understand what I mean.

Thanks

kilrah, 2017-02-25:
Please detail whether you're referring to companion or radio. Current "next" status the radio do allow Custom subprotocols up to 63, but Custom doesn't even exist in companion.

profdrdenis, 2017-02-25:
Hello,
before we talk wrong I talk from protocol not from Subprotocol. In Companion I count and have 29 last is "???" 

I start my topic about radio.

schwabe, 2017-02-25:
@profdrdenis that "???" will be fixed by https://github.com/opentx/opentx/pull/4459

kilrah, 2017-02-25:
> I start my topic about radio.

Then that's already fixed for next release.

closed 2017-02-25 by kilrah