opentx/opentx #4508: Improve Horus graphics speed

projectkk2glider, 2017-02-25:
Currently `lcd.drawFilledRectangle(0, 0, LCD_W, LCD_H, TEXT_BGCOLOR)` from Lua takes 80ms. I see why (each pixel is drawn one by one, and for each one the check is done if it is withing bounds of the screen).

The plan:
 * add graphics benchmark test to CLI and use it to get the baseline speed.
 * tweak the graphics code to get better performance and compare results with the benchmark each time.

Optimization ideas:
 * move `are we in the buffer` check higher up the funcitons - much less calls to the check.
 * use DMA whenever possible (like draw solid filled rectangle above)

projectkk2glider, 2017-02-25:
Some work done in https://github.com/opentx/opentx/tree/projectkk2glider/graphics_performance2 , results:

Baseline in 6dffebd4ac531a9027bb1fe2bb8c123b83449781

```
Starting graphics performance test...
Test testEmpty speed: 1804100.12, (1804100 runs in 1000 ms)
Test testDrawSolidHorizontalLine2 speed: 7910.00, (7910 runs in 1000 ms)
Test testDrawSolidVerticalLine2 speed: 3433.13, (3440 runs in 1002 ms)
Test testDrawDiagonalLine speed: 428.29, (430 runs in 1004 ms)
Test testDrawSolidFilledRectangle speed: 1.58, (10 runs in 6346 ms)
Test testDrawSolidFilledRoundedRectangle speed: 6.29, (10 runs in 1590 ms)
Test testDrawRect speed: 66.29, (70 runs in 1056 ms)
Test testDrawFilledRectangle speed: 3.00, (10 runs in 3330 ms)
Test testDrawBlackOverlay speed: 0.96, (10 runs in 10368 ms)
Test testDrawText speed: 46.95, (60 runs in 1278 ms)
Test testClear speed: 33.90, (40 runs in 1180 ms)
Total speed: 11930.39

Starting graphics performance test...
Test testEmpty speed: 1814930.12, (1814930 runs in 1000 ms)
Test testDrawSolidHorizontalLine2 speed: 7900.00, (7900 runs in 1000 ms)
Test testDrawSolidVerticalLine2 speed: 3433.13, (3440 runs in 1002 ms)
Test testDrawDiagonalLine speed: 428.29, (430 runs in 1004 ms)
Test testDrawSolidFilledRectangle speed: 1.58, (10 runs in 6346 ms)
Test testDrawSolidFilledRoundedRectangle speed: 6.29, (10 runs in 1590 ms)
Test testDrawRect speed: 66.29, (70 runs in 1056 ms)
Test testDrawFilledRectangle speed: 3.00, (10 runs in 3330 ms)
Test testDrawBlackOverlay speed: 0.96, (10 runs in 10368 ms)
Test testDrawText speed: 46.95, (60 runs in 1278 ms)
Test testClear speed: 33.90, (40 runs in 1180 ms)
Total speed: 11920.39
```

After first round of optimizations in e7b54efdc7e85e32572bbe4b68bb49d7bc116cd0

```
Starting graphics performance test...
Test testEmpty speed: 1814910.12, (1814910 runs in 1000 ms)
Test testDrawSolidHorizontalLine2 speed: 8030.00, (8030 runs in 1000 ms)
Test testDrawSolidVerticalLine2 speed: 3450.00, (3450 runs in 1000 ms)
Test testDrawDiagonalLine speed: 428.29, (430 runs in 1004 ms)
Test testDrawSolidFilledRectangle speed: 33.90, (40 runs in 1180 ms)          21 x SPEEDUP
Test testDrawSolidFilledRoundedRectangle speed: 133.84, (140 runs in 1046 ms) 21 x SPEEDUP
Test testDrawRect speed: 588.82, (590 runs in 1002 ms)                         9 x SPEEDUP
Test testDrawFilledRectangle speed: 3.96, (10 runs in 2528 ms)              1.32 x SPEEDUP
Test testDrawBlackOverlay speed: 1.20, (10 runs in 8332 ms)                 1.25 x SPEEDUP
Test testDrawText speed: 52.63, (60 runs in 1140 ms)
Test testClear speed: 33.90, (40 runs in 1180 ms)
Total speed: 12756.54

Starting graphics performance test...
Test testEmpty speed: 1803890.12, (1803890 runs in 1000 ms)
Test testDrawSolidHorizontalLine2 speed: 8030.00, (8030 runs in 1000 ms)
Test testDrawSolidVerticalLine2 speed: 3443.11, (3450 runs in 1002 ms)
Test testDrawDiagonalLine speed: 428.29, (430 runs in 1004 ms)
Test testDrawSolidFilledRectangle speed: 33.90, (40 runs in 1180 ms)
Test testDrawSolidFilledRoundedRectangle speed: 133.84, (140 runs in 1046 ms)
Test testDrawRect speed: 587.65, (590 runs in 1004 ms)
Test testDrawFilledRectangle speed: 3.96, (10 runs in 2528 ms)
Test testDrawBlackOverlay speed: 1.20, (10 runs in 8332 ms)
Test testDrawText speed: 52.54, (60 runs in 1142 ms)
Test testClear speed: 33.90, (40 runs in 1180 ms)
Total speed: 12748.39
```

projectkk2glider, 2017-02-25:
Another data point: Lua S6R script on first page (no receiver connected), the time for one loop of the script:
 * 6dffebd LUA fg   : 80ms - 100ms
 * e7b54ef LUA fg   : 9.436ms - 15.496ms

Which brings us to the something I quite don't understand:

 Most of the time above in first case is spent by drawing white background over the whole LCD `lcd.drawFilledRectangle(0, 0, LCD_W, LCD_H, TEXT_BGCOLOR)`.  Removing this line reduces the run time for ~80ms.

That would suggest that one fill of the whole LCD screen takes 80ms. **But** according to the graphics performance test we have `testDrawSolidFilledRectangle speed: 1.58, (10 runs in 6346 ms)` (this test just calls `lcdDrawFilledRect(0, 0, LCD_W, LCD_H, SOLID, TEXT_BGCOLOR);` which is the same as in script)

But during the graphics test each run of the `lcdDrawFilledRect` takes 600ms! So why is the time in the test so much bigger?

projectkk2glider, 2017-02-26:
Another finding: setting the whole display buffer to black (lcdClear()) takes 30ms with both of these methods:
 * `memset(data, 0, LCD_W*LCD_H*2);`
 * `DMAFillRect(data, width, height, x, y, w, h, lcdColorTable[COLOR_IDX(flags)]);`

The fact is that we need to set the value of 480×272×2 = 255kB! ob memory locations.

So the conclusion (which is kind of logical) is that DMA is as fast as direct memory access. The limit here is relatively slow access time to external SDRAM. Other unused feature is that DMA could run in background and other stuff could be done while it is running, but actually using this will be tricky.

Can we overclock the SDRAM and/or are we even even using the max allowed speed now?

mpaperno, 2017-02-28:
I've been giving some thought to the timing issues and my first thought with the benchmarks is whether the timer is really accurate in this case.  Also that milliseconds is a rather "long" measure for precise timing (though of course aggregating the results helps mitigate that).  Not having a Horus though, I can't verify or test any of these theories.

What I do have is some code which sets up a microsecond timer w/out any interrupts (using 32-bit TIM5).  I used this while [benchmarking](https://github.com/opentx/opentx/issues/3318#issuecomment-264688157) the Lua compiler changes.  Would that be helpful?  (We use this kind of timer in Autoquad flight controller for precise measurements, eg: https://github.com/mpaperno/aq_flight_control/blob/next/src/aq_timer.c but w/out the alarms/interrupts.)

It will require some small changes for Horus because TIM5 right now is used on rev 13+ for backlight, but I'm pretty sure that can be moved to TIM9 (shared with haptic driver, but different channel).  PA03 is TIM9_CH2.

mpaperno, 2017-02-28:
I'm not even sure what "systick" represents right now... https://github.com/opentx/opentx/blob/next/radio/src/thirdparty/CoOS/OsConfig.h#L88    Kinda looks like 2ms.  https://github.com/opentx/opentx/blob/next/radio/src/thirdparty/CoOS/portable/OsArch.h#L48   

AQ uses 1KHz here which has always resolved to 1ms... https://github.com/mpaperno/aq_flight_control/blob/next/src/co_os/OsConfig.h#L50

projectkk2glider, 2017-03-01:
We already have pretty god timer system for debugging, check out macro `DEBUG_TIMER_START` and associated CLI command `p dt`.

projectkk2glider, 2017-03-01:
I had another *breakthrough*, for transparency masks I will use DMA2D A4 bitmap format (two pixels packet into one byte). This can be used for fonts and for all graphics masks. There are several benefits:
 * fast
 * we can drop fonts cache and use fonts directly from FLASH (unless reading from FLASH is much slower than SDRAM).
 * less memory used for masks (4x times less) while maintaining the same graphics quality.

mpaperno, 2017-03-01:
Then why are you using CoGetOSTime for timing?  https://github.com/opentx/opentx/commit/6dffebd4ac531a9027bb1fe2bb8c123b83449781#diff-d0bc8fd5de94ee8a3e7d3dc436422c83R422

TIM5 is 32-bit and would always have micros resolution for any task < ~71min.  Simple, fast, no interrupts, no math.  Last thing you want in a precise timer is any additional overhead.  (Why use a 2MHz/.5us timer and then divide by 2 to get 1us resolution?)

You have a "ton" of unused SRAM on the Horus, all accessible via DMA.  Wouldn't that be the fastest place/method to read from?  How much memory do you need?

projectkk2glider, 2017-03-01:
I use CoGetOSTime to time the whole test run which is usually around 1s so 2ms resolution here is not a problem.

We are using SDRAM for graphics! see https://github.com/opentx/opentx/issues/4508#issuecomment-282539565

projectkk2glider, 2017-03-01:
No, every malloc() comes from SDRAM.

projectkk2glider, 2017-03-02:
I am reading http://www.st.com/resource/en/application_note/dm00287603.pdf (LCD-TFT display controller (LTDC) on STM32 MCUs)

Here is what we got now:
 * LCD Pixel clock is 16Mhz
 * total pixel count is 525x286 which results in a refresh rate of 106Hz
 * for each frame LCD needs to read 2 x 255KB (two layers)
 * so effective read rate from SDRAM to LCD is 106 x 2 x 255KB => 54MB/s

I don't know what it theoretical max read rate for our SDRAM, but this figure looks big. Remember that when LCD is reading SDRAM no other device can read/write it. So everything that we have that uses SDRAM must share (Lua, graphics).

The idea is to reduce the bandwidth used by LCD controller, this is how:
 * reduce the refresh rate (if it is allowed by the TFT itself) or increase VBP (vertical blanking period)
 * use only one layer (rate halves), we don't really use two layers effectively, the double buffering will still be used.

With some of the SDRAM available bandwidth now freed, other stuff should work faster (like preparing graphics for display, SD card reading (disk cache)).

Further optimizations:
* the two LCD buffers in SDRAM should be placed in different banks (but make sure we don't loose much space)

 >The external SDRAM is composed of multiple banks. Given that making random accesses
on a bank generates some precharge and activates some cycles, the framebuffer must be
placed in an independent bank accessed only by the LTDC. This action reduces the external
memory latency and leads to a higher throughput. As a consequence, when double
framebuffer technique is used, it is recommended to have these buffers in two separate
banks.

* possibly enable SDRAM_READBURST (which is disabled now) but need to check what are possible negative consequences.
>Another interesting feature allowing to optimize the reading performances from the SDRAM
is the use of RBURST.
>The SDRAM controller adds a cacheable read FIFO with a depth of 6 32-bit lines. The read
FIFO is used when the read burst is enabled and allows to anticipate the next read
accesses during CAS latencies.

projectkk2glider, 2017-03-03:
Due to a bug in my testing code all reported speeds are 10x to small :stuck_out_tongue_closed_eyes:  This are the results of some not yet committed state:
```
Starting graphics performance test...
Test testEmpty speed: 17876002.00, (17876000 runs in 1000 ms)
Test testDrawSolidHorizontalLine2 speed: 77900.00, (77900 runs in 1000 ms)
Test testDrawSolidVerticalLine2 speed: 33000.00, (33000 runs in 1000 ms)
Test testDrawDiagonalLine speed: 4282.87, (4300 runs in 1004 ms)
Test testDrawSolidFilledRectangle speed: 338.98, (400 runs in 1180 ms)
Test testDrawSolidFilledRoundedRectangle speed: 990.10, (1000 runs in 1010 ms)
Test testDrawRect speed: 5917.16, (6000 runs in 1014 ms)
Test testDrawFilledRectangle speed: 39.25, (100 runs in 2548 ms)
Test testDrawBlackOverlay speed: 88.65, (100 runs in 1128 ms)
Test testDrawText speed: 522.65, (600 runs in 1148 ms)
Test testClear speed: 338.98, (400 runs in 1180 ms)
Total speed: 123418.66
```

projectkk2glider, 2017-03-03:
My ramblings about the SDRAM and LCD above are right. Here are the results of the memory speed test with current LCD configuration:
```
Starting memory speed test...
Test testMemoryReadFrom_RAM_8bit speed: 6796.12, (1400 runs in 206 ms)
Test testMemoryReadFrom_RAM_32bit speed: 27000.00, (5400 runs in 200 ms)
Test testMemoryReadFrom_SDRAM_8bit speed: 1428.57, (300 runs in 210 ms)
Test testMemoryReadFrom_SDRAM_32bit speed: 5092.59, (1100 runs in 216 ms)
Test testMemoryCopyFrom_RAM_to_SDRAM_8bit speed: 4500.00, (900 runs in 200 ms)
Test testMemoryCopyFrom_RAM_to_SDRAM_32bit speed: 14705.88, (3000 runs in 204 ms)
Test testMemoryCopyFrom_SDRAM_to_SDRAM_8bit speed: 5092.59, (1100 runs in 216 ms)
Test testMemoryCopyFrom_SDRAM_to_SDRAM_32bit speed: 13500.00, (2700 runs in 200 ms)
Disabling LCD...
Test testMemoryReadFrom_RAM_8bit speed: 6796.12, (1400 runs in 206 ms)
Test testMemoryReadFrom_RAM_32bit speed: 27000.00, (5400 runs in 200 ms)
Test testMemoryReadFrom_SDRAM_8bit speed: 2521.01, (600 runs in 238 ms)
Test testMemoryReadFrom_SDRAM_32bit speed: 8910.89, (1800 runs in 202 ms)
Test testMemoryCopyFrom_RAM_to_SDRAM_8bit speed: 4500.00, (900 runs in 200 ms)
Test testMemoryCopyFrom_RAM_to_SDRAM_32bit speed: 14705.88, (3000 runs in 204 ms)
Test testMemoryCopyFrom_SDRAM_to_SDRAM_8bit speed: 5092.59, (1100 runs in 216 ms)
Test testMemoryCopyFrom_SDRAM_to_SDRAM_32bit speed: 13500.00, (2700 runs in 200 ms)
Total speed: 161142.28
```

First part is run with the LCD running (reading from SDRAM), second part with LCD stopped. As expected speeds increased in second part. 

The strange part is the memory copy test, it is not influenced by the LCD and is actually faster then just reading from SDRAM!? 

testMemoryCopyFrom_SDRAM_to_SDRAM_32bit uses this code to copy 4kB blocks:
```c
void memoryCopy(uint32_t * dest, const uint32_t * src, uint32_t size)
{
  while(size--) {
    *dest = *src;
    ++src;
    ++dest;
  }
}
```

Assembly of the function above:
```asm
509:/opentx/radio/src/cli.cpp ****   while(size--) {
.L288:
  cmp r3, r2  @ ivtmp.408, size
  beq .L290 @,
.L289:
510:/opentx/radio/src/cli.cpp ****     *dest = *src;
  ldr r4, [r1, r3, lsl #2]  @ D.28203, MEM[base: src_5(D), index: D.29478_17, offset: 0B]
  str r4, [r0, r3, lsl #2]  @ D.28203, MEM[base: dest_4(D), index: D.29478_17, offset: 0B]
  adds  r3, r3, #1  @ ivtmp.408, ivtmp.408,
  b .L288 @

```

While the `testMemoryReadFrom_SDRAM_32bit` (which should be faster in my opinion) uses this:
```c
void memoryRead(const uint32_t * src, uint32_t size)
{
  while(size--) {
    *(const uint32_t volatile *)src;
    ++src;
  }
}
```
```asm
.L272:
490:/opentx/radio/src/cli.cpp ****   while(size--) {
  adds  r1, r1, #-1 @ size, size,
.LVL433:
  bcc .L274 @,
 491:/opentx/radio/src/cli.cpp ****     *(const uint32_t volatile *)src;
  ldr r3, [r0], #4  @ vol.81, MEM[(const uint32_t *)src_1]
  b .L272 @
```

projectkk2glider, 2017-03-14:
Some more resources:
 * explanation of LCD controller and Chrome-ART (DMA2D) accelerator http://www.compel.ru/wordpress/wp-content/uploads/2013/11/1_LTDC_ChromeART.pdf

closed 2020-04-06 by stale[bot]

Luznatural, 2021-02-19:
Any news from this issue?