opentx/opentx #4509: Companion 2.2rc10 for Mac Taranis Telemetry Simulator window on top or not

athertop, 2017-02-25:
Telemetry simulator - panel opens on top and stays on top, until you click back on the main simulator screen, then the telemetry simulator panel gets stuck behind the other panel and no way to bring it forwards again.

Regardless of whether the Telemetry simulator panel is at the front or stuck behind the telemetry panel, switching to a different app in OS X leaves this telemetry simulator panel in front of the new app, blocking ut from view - same for all apps. Only way to get rid of the simulator panel from blocking the screen is to close it.

kilrah, 2017-02-25:
Next RC will have a completely new UI for the simulator.

closed 2017-02-25 by kilrah

athertop, 2017-05-18:
This same issue still exists even in the latest release. See here:
[image]
Notice how the Telemetry simulator window gets stuck behind the Radio Simulator window on a Mac (not sure how this behaves on Windows/Linux sorry). Strangely, if you switch focus to a doifferent app, then switch back with Cmd-Tab the Telemetry window is once again at the front. I know this is just a cosmetic issue, but wondering why it just can't be made to behave like any other window?