opentx/opentx #4512: companion 2.2 doesn't recognise 2.1 model backups, cant exctract radio BIN.

lshems, 2017-02-26:
direct loading of a backup model 2.1 leads to error.

Loading of backup model in existing open file in Companion results in nothing.

lshems, 2017-03-03:
I personally think that exchanging EPEE's is not the correct way to share model setups. It has been happening to often that people don't copy the models from the source epee to their own eepe as destination, but simply copy the entire EPEE to the transmitter, with all consequences that you know better than I do.

Why not use the backup model and restore model where they where intended for? You can even exchange models in the field in that way. Isn't that what should be promoted as way to exchange setups?

Well, it isn't going to help if the backup and restore functions aren't working properly from within companion.

I think it is good practice to have models backupped for restore and archiving reasons (not loosing my setups). If I cant restore them in an upgraded version of OpenTX, then I run into a lot of trouble and frustrations.

Not that my frustration is your concern, but I think this is a generic concern: happy users?

kilrah, 2017-03-03:
Model backups are only meant to be used on-radio, and on the same radio with same version and same options.

They can't be reloaded on a different setup since they depend on the radio settings they were made with. That's the whole reason for an eepe (now OTX) to include everything.

I also backup and archive my setups, and backing up the whole eepe ensures I always have the full state of things I can refer to and/or copy if necessary. It's not like opening 2 documents and dragging a model is complicated.

schwabe, 2017-03-03:
A reason why people might try that is that frsky lists "Model configuration files compatible with TARANIS X9D Plus" as feature for the X7

kilrah, 2017-03-03:
"When going through companion conversion with the whole EEPROM", which they don't specify nor even distribute companion ;)

lshems, 2017-03-03:
"Model backups are only meant to be used on-radio, and on the same radio with same version and same options"

So you tell me that the possibility to exchange models in the field by swapping SDcards is not meant for that purpose??

It is the only way to do it without a PC i think, other than making an eeprom backup, load another eeprom, backup the models from that eeprom, load my own eeprom again, and restore the models I just backupped.

Ok, nice. Not my intuitive feeling of how this should work.

But do whatever you like, it's your system. Always surprised about the argumentation though. But that must be just me.

3djc, 2017-03-03:
Indeed this is the case, since models may rely on radio settings, so a model alone is not something that can be used for transfer anyway. It says backup, not 'pack for transfers' or something like that.

Would it be nice that models where not radio aware and easy to transfers without any constrains ? Sure it would, but it needs a very very important change to the whole structure, and could not happen anyway before a 3.x version or something like that (and would probably not work on existing taranis due to memory constrains). Is it something we have in mind and pursue, yes sure, but it's not for tomorow

kilrah, 2017-03-03:
> So you tell me that the possibility to exchange models in the field by swapping SDcards is not meant for that purpose??

Not at all, it was only made for people who had tons of models and were tight with the 60 slots. That way they can "unload" some to free space and "reload" another. Same for backing up a model before change, us "mortals" jsut make a copy but if you're tight and don't want multiple copies the Backup function is useful.

Conversions between radios need companion. It's already difficult enough to get right and keep in sync with the unlimited resources of a computer, on the radio it would be unmanageable.

lshems, 2017-03-03:
Well, at least that is some argumentation I can understand. It still isn't the intuitive way to do things, but if there are restrictions, there are restrictions. But this makes me wonder:

"Indeed this is the case, since models may rely on radio settings, so a model alone is not something that can be used for transfer anyway. It says backup, not 'pack for transfers' or something like that."

what radio settings are you referring to? I have no clue to what these could be. Thanks for any clarification.

kilrah, 2017-03-03:
Everything in the Radio settings menu (General settings in companion). That includes which switches and pots are installed and what type if any.

lshems, 2017-03-03:
OK thanks. But do you say that my backup will be different if I have a different language selected, or a different channel order? I would assume that these are not part of the backup, just for reasons that they are radio wide (and thus the part that IS there in the EPEE to make sure you backup all the radio)???

If I do not have any pot installed, that doesn't make my backup different, does it. It is a radio setting, not a model setting.

Confused ?!

lshems, 2017-03-03:
Also, if I make a backup, and change my radio settings, I'm not supposed to restore? It becomes really fuzzy  for me to understand all this.

projectkk2glider, 2017-03-03:
Your restored backup will use whatever radio settings it finds. If you for example changed physical switch configuration in the meantime the backup will still happily use the same switches which could now be configured differently.

kilrah, 2017-03-03:
The Hardware page is the important one. For example Voice language would cause any custom sounds not to work anymore unless you have a copy of those in the appropriate language folder, but that's not "critical".

3djc, 2017-03-03:
To some extend, it is like backing up a part of a pc. If nothing else changed, you can restore it, if not, well expect at least major problems, if possible at all. Having a partial backup doesn't mean you can restore it no matter what

lshems, 2017-03-03:
OK, but obviously if I change some radio settings, those will be reflected on my radio behaviour.

So, ok, if I'm on open 2.0 or 2.2, we don't have any hardware settings, do we? apart from calibration.

On 2.2 we have hardware settings, but they won't affect the model backup content. The backup is still the same, no matter what the radio settings are.

Well, I do not see where there is a problem then to use the backups and restore them, because dragging a model from one EPEE to another on companion does exactly the same, does it?

So, no matter how you formulate this, it seems the intuitive way to transfer models is by backup and restore.

Not because I prefer, but because that is how it works on all equipment using setting files as far as I know, and there is essentially no difference when using drag and drop from one epee to another.

If there is, I would really like to understand what that would be.

Thanks.

kilrah, 2017-03-03:
> So, ok, if I'm on open 2.0 or 2.1, we don't have any hardware settings, do we? 

They were introduced in 2.1.

> and there is essentially no difference when using drag and drop from one epee to another.

Yes companion has the ability to do conversions while doing such transfers, or at least mark the invalid settings and prevent you from writing to a radio with incompatible settings until you've corrected them. The facilty was just added so there aren't many cases handled yet but more should come. This won't be added to the radio though.

lshems, 2017-03-03:
Still, that doesn't answer the question. Are there any differences in the content of the backuped model, due to radio settings?

This question is really important for all users.

That companion can warn you that a switch is used in the model that is not configured on the EPEE you are dragging it to, sure, that may be useful and practical.
But I think you are relying to much on people having access to a computer to change their transmitter, or fiddel around with settings and models. 

People do this in the field all the time, and if this would become impossible, that would be an enormous drawback of openTX. 
With the shortening of names and labels, we are already having trouble keeping track of how the model setup is to be changed while on the field and in a hurry, during competition.

So, would it be possible to confirm that the radio setup settings are in no way influencing the content of the backuped model file?

Thanks.

P.S. sorry if I seem to be nagging here, but I think understanding how backup and restore features work is essential for everyone, not only to me.

kilrah, 2017-03-03:
> Still, that doesn't answer the question. Are there any differences in the content of the backuped model, due to radio settings?

The contents of the backed up model do not differ, but those contents may not be valid when loaded on a radio with different radio settings. Behavior in this case is undefined, and figuring out which settings are invalid is not easy since you need to go edit every occurrence to even see it.

lshems, 2017-03-03:
Ok, thanks for the clear answer. Comforting, really. 

Well, if you wont implement import export for older versions would be a pity, but understandable due to time or whatever restrictions.

Supporting it for the same version would be necessary in my opinion.

kilrah, 2017-03-03:
> Supporting it for the same version would be necessary in my opinion.

It simply isn't possible. Even if we included a copy of the hardware settings in the model so that the radio that "restores" could detect they are different, it couldn't do anything but block you from restoring it.

Any "auto-fixing" would mean attempting to guess what the user wants, which is never good and we'll never do since we'll never pretend we know what he wants.

A simple and common case, someone has installed a multipos switch on their radio. The model depends on the position of these switches. Trying to import that on a radio with no multipos switch will require extensive manual work, not only to change the switches but probably to configure several new logical switches to fit the recipient's layout and desires.

lshems, 2017-03-03:
Sorry, I was not clear in what I meant, perhaps.

I think it should be logical to be able to open a backup model made on a 2.2 radio in companion 2.2. That is how it works for 2.1 as well.
That is also what a drag and drop does: loading the MODEL settings from one EPEE and trying to paste them into another EPEE. 

Then some magical logic is performed, to warn you of impossible settings. If yoy get some warnings, you would have to adapt your existing model in the source EPEE, then do the drag and drop, and restore or adapt the new model to fit the new hardware settings, and restore manually whatever you have deleted or changed in the first model.
That is at least how I understand your explanation.

OK, since you manage the trouble of you having to do all this magical checking, I would be happy with that. :)

But why shouldn't I be able to open the backup of that same model and have it checked with the same logic before adding it to the epee I am trying to add it to? Seems perfectly logical to me.

Then, same thing, why would I want to bother writing an entire eeprom to the radio if I just want to add one model to the existing one on my friends radio? So exporting a single model from an Epee would be logical as well, to be able to import it on the radio. There are some people who don't want me to hook up their radio to my companion, because they have had bad experiences of people loading another EPEE on their radio. Restoring a model from a backup file is perfectly acceptable for them.

If I try to open a backuped model, why not try to paste it in the default EPEE, that is created if I make a new one in companion?
That is happening in the OpenTX 2.1 version, if I understand it properly.

Really, I still don't understand your arguments of not supporting that (the option is already there in RC10, just not working).

I cannot think of any difference between drag and drop and backup and restore. Neither between radios, nor between EPEE's (those are effectively the same anyway).
Backup and restore should work exactly the same as copy and paste. If not, we have a big problem of understanding these concepts, as they would then differ from the common notion of backup and restore.

Anyway, I can work my way around all that. I don't need it really. 
But I reckon it's not only me that's counting here.

kilrah, 2017-03-03:
> I think it should be logical to be able to open a backup model made on a 2.2 radio in companion 2.2. That is how it works for 2.1 as well.

Because as I mentioned somewhere we can't even know what radio this backup comes from to do the required conversions. When you use the full eepe we do know that, and the whole is converted to the appropriate format when you open it or change profile before you start copying things. On 2.1 it was OK due to the similarity of the supported radios but now we have radios that are too different to assume the result will be decent.

> Then, same thing, why would I want to bother writing an entire eeprom to the radio if I just want to add one model to the existing one on my friends radio?

Because that's how the underlying filesystem works. There's no reason you'd want to, but there are reasons why you have to.

> I cannot think of any difference between drag and drop and backup and restore. Neither between radios, nor between EPEE's (those are effectively the same anyway).

See first point.

lshems, 2017-03-03:
Ok, fair enough, I see that for some reason or another you didn't store any info on the model type into the backup file. Still the version differnce should be clear, since the structure will be different for 2.2 and 2.1or 2.0.

So why not have the import of a backup file check the opentx version, stop with a warning if it's an older version, and have a warning pop up that model backup can only be restored properly if they were created with the same radio type (x9->x9; x7->x7, etc).

This would be a reasonable compromise, leaving the already active menu option on the epee (right click and you get that option) to restore a backup in place. Ctreating a local backup can be done from the radio simulator, so no problems there, just inconvenience.

?

3djc, 2017-03-03:
Thats where the catch comes in ! It is NOT a structured file with fields, separators and things like that. It is a 'brutal' dump of a memory segment (remember how ram is tight on those radio). I'll try to oversimplify  to illustrate. Let's say you have a byte like : 01 01 01 01, it could be a status for eight dual position switch, or 4 dual switch and 2 three position, there is nothing telling you which one it is, only the combination of otx version/radio type/radio settings will work as planned with it, all other will produce unexpected result, but there is no way to tell for what combination they where extracted from.

What we might (and I stress the might) be able to do is check the total size, and compare to what current config would produce, but that would not catch all scenarios, by far (because some have the same size, but different meaning)

lshems, 2017-03-03:
Could including a header as of 2.2 be an option? almost every file starts with a preamble to make it recognisable. Even a lua compiled file starts with LUAX, where X denotes Capital letter corresponding to the mayor version.

Just thinking about a solution.

It's not that we want to fix a thing of the past. :)

kilrah, 2017-03-03:
No, too late, would have to wait for 2.3.

stanislavdimov, 2017-03-05:
OpenTX Companion 2.2.0 RC11 (N363) - Windows Installer  crash my model after write on radio frsky taranis+

kilrah, 2017-03-05:
Firstly this is completely unrelated, you should create a new issue. Secondly "Windows Installer crash my model" means nothing at all. Please include more information.

flaviomartins8, 2018-03-02:
My radio today Horus X10s today, change sys color to black , the first screen was calibration in black color and can't extract RADIO/radio.bin  occurs.
I reinstall everything and happens again.  firmware, Sd format and lose my models 
I can't open the backup two.
Any news?

kilrah, 2018-03-02:
Sounds like an SD card issue. Hard to comment without more details, version, ... but what's sure is that it's not related to this issue nor an OpenTX bug...

closed 2019-11-15 by stale[bot]