opentx/opentx #4514: [MULTI] had support for more than 31 protocols

pascallanger, 2017-02-26:
Hi,
I got report that opentx only supports 31 rf protocols where the serial protocol supports 63. Er9x and ersky9x already supports this for some time.
Basically, the custom entry does not go over 31. I've published a new RF protocol gw008 number 32 a few days ago and opentx users can't access it.
Pascal

schwabe, 2017-02-26:
Hm, 

it should support it. Maybe the users are using very old 2.1.x versions?

[image]

pascallanger, 2017-02-26:
OK so feel free to close this false issue.
Thanks for checking, Pascal

closed 2017-02-26 by pascallanger

pascallanger, 2017-02-26:
By the way have you tried to load the latest multi code and load the protocol gw008 32? It should blink forever if no quad is around. I think it's a good test to see if it works as it should.
Pascal

powercycle, 2017-02-26:
I can confirm that custom choice goes to 63, and at protocol 32 for multi 1.1.6 using RC10, shows Protocol Invalid. 
As of today, the latest precompiled release is 1.1.6, so folks may not be recompiling from master.