opentx/opentx #4518: Copying multiple mix lines

mpjf01, 2017-02-26:
The relatively new S6R and S8R receivers do, for the first time , enforce a specific channel order. In order to use the new receivers with existing model definitions - either your own or those produced by others - requires rearrangement of the channels unless you happen to be following the AETRAFF convention already.

Testing a S8R in a 6 servo glider required me to reallocate the channels for 55 mixes, each of which had to be copied and pasted individually, and it occurred to me that an ability to copy a whole channel's worth of mixes at a time would be useful, as would the relatively standard mechanisms for making multiple selections for copy/delete operations (Shift/Control). Has any consideration been given to this in the past?

projectkk2glider, 2017-02-27:
There was an idea floating around a while back to have a new feature added to OpenTX that would enable user to map mixer channels to receiver channels. Usually the mapping would be 1-1 so mixer 1 would output to CH1, mixer2 to CH2, etc.. (like it is now). Then user could for example select for mixer2 to be mapped to CH7.

It was concluded that this, while useful and very flexible, would add more harm than good - it would confuse users and make support problems much worse.

Now I had one more thought about this and I have perhaps found a way to include this, but not confuse anyone (too much). Here is my idea:
 * implement mixer to output mapping (probably requires eeprom change) by default it is **turned off** or disabled
 * when the new feature is disabled it is not visible anywhere in the GUI - no confusing of unsuspecting users.
 * user has to enable it for each model separately, when enabled a new GUI setting (either a new page or on some existing page) will be available that will enable user to map outputs.
 * the mapping would be done at the very end (after output limits) - this would have the same effect as swapping servo connectors on a dummy receiver (the one that does not care about channel ordering like X6R, D8, etc).
 * The existing displays of mixer and output monitors would not change regardless of mapping - the change will only be visible to the values sent to TX module.

I think this would be good, because:
 * users that don't need this will never see it
 * users that have use for this will be aware of it and will know how to use it.
 * this feature goes along nicely with the whole concept of OpenTX - highly configurable.
 * this feature would immensely simplify sharing of existing complex templates (rcsoar for example) that use extensive mixing. User the just needs to map the channels to suit his particular servo/receiver configuration.

Please comment....

strgaltdel, 2017-02-27:
.. that would be a very nice feature for the more advanced users.

Some month's ago i would have needed this functionality for configuring a 6flap glider, based on an 4flap template, when i ran into the limit of maximal usable mixer lines.
This caused a lot of new arrangements in the mixers..

My thought was if it would be able to implement something with the functionality of a "patch panel" like used in network infrastructure (logically seen between mixer and output block).
There you can connect sources (switch ports, pbx ports..) to corresponding consumer ports which can vary in location and protocol  (phones, printer, PC's).
So you would have a free choice which mixer channel would be logical connected to which output.

As a second effort the interchangeability of templates would increase.
Often colleaques request templates from another, but cause they've an ready to fly model with difficult access to the Rx, they need to reorganize the mixer lines.
(which often causes some failures).

Your approach covers this needs by 100%, so, from my point of view, it would be nice to see it implemented in oTx !

regards
Udo

mpaperno, 2017-02-27:
The proposal makes a lot of sense to me.  This is essentially how Multiplex does (did?) it, eg. on the Profi/Evo/etc.  Mixes are not tied to any outputs per-se.  You pick a mix and then assign it to any output channel (they call them "servos").  On a separate screen you can set limits/curves/expo/centers/etc per "servo."

mpaperno, 2017-02-27:
> On a separate screen you can set limits/curves/expo/centers/etc per "servo."

I should probably have specified that these would be in addition to any curves/expos/etc already set on the actual "mix" or "control" (MPXs versions of OTx Mixes & Inputs).

elecpower, 2017-02-27:
+1
Agree it does add more potential complexity but at the same time more flexibility. But we already have some advanced configs eg rcsoar's. It only expands on the Inputs and Mixes freedom we have now. 
Though not in favour of hiding the mix link to the output channel in the GUI.There is currently no hiding the other flexibility eg real and virtual inputs, real and virtual mixes/channels with hierarchic linking.
The Output channels would likely have a field to link to the final mixer line so cannot see how that would be more confusing.
The new model wizard gets those new to OpenTX started.

elecpower, 2017-02-28:
The more I think about it the more I like the idea. It could actually improve understanding of the flow from physical inputs signals to physical output signals. Mixes would become a calculation engine not even needing to be aware of outputs.
Currently it is not necessary for all Output channels to be transmitted to the receiver and not even from channel 1 (not a criticism). Also the Output transmitted channel block needs to be contiguous from the start channel number which has a flow on back to the need for Mixes sequence to match receiver channels.
It decouples mixes and any references to virtual channels instead mixes would reference other mixes not channels.

Allerhopp, 2017-02-28:
+1
The OpenTX task is similar to a Audio DSP and there is also a completely free (graphic) routing to the outputs. Someday we will have the same procedure for TX programming - it is a logic step
[image]
.

ja-jaa-org-uk, 2017-03-01:
+1
I fully support the decoupling of mixers and output channels
I have recently been updating my planes for the Horus and have found the lack of such
a facility a great drawback in the development of a versatile template.
https://www.rcgroups.com/forums/showpost.php?p=36986432&postcount=8586
The extra level of abstraction seems a very small price to pay for the additional fexibility

strgaltdel, 2020-01-07:
Ich kehre zurück am 08.01.2020.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [opentx/opentx] Copying
multiple mix lines (#4518)" gesendet am 07.01.2020 14:00:10.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

closed 2020-10-03 by stale[bot]

strgaltdel, 2020-10-03:
Ich kehre zurück am 05.10.2020.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [opentx/opentx] Copying
multiple mix lines (#4518)" gesendet am 03.10.2020 16:34:15.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.