opentx/opentx #4521: Taranis 9E OpenTx 2.2 RC10 (N362) problem with using USB port when tx is powered.

Gunnergolly, 2017-02-27:
Have just installed 2.2 RC10 on my Taranis 9E and did a quick test with my neXt helicopter PC simulator to see if things were working ok with the new build with the sim. The Tx is already powered up with the model selected and as soon as i plug the USB cable in it causes the 9E to lock up on whichever screen i had displaying at the time. 

Pressing the buttons on the Tx have no effect, however if say i was on the Tx stick position screen  at the time i plugged the USB plug in i can see the two joystick displays emulating the position of my sticks as i move them around.

The "neXt simulator set up/calibration page where i can monitor the output from my Tx is not registering anything so there looks to be a problem with the USB side of things using RC10.

Unpluging the USB cable causes the 9E to reboot, have also tried with the 9D+ and exactly the same thing happens.

Is this a known issue or is there a workaround ?.

Thanks.

kilrah, 2017-02-27:
Sounds like you flashed firmware with the "massstorage" option selected, which replaces the joystick functionality with SD card access.

closed 2017-02-27 by kilrah

Gunnergolly, 2017-02-27:
Yes i did, thanks for your reply.