opentx/opentx #4525: In 2.1.9, why is MULTIPLY allowed in the first mix line?

Miami-Mike, 2017-02-28:
It seem to me that MULTIPLY should automatically be changed to ADD when it winds up in the first mix line of a channel, since multiplying something by zero can never produce a useful result.

Am I missing something, or has it already been changed for 2.2?

kilrah, 2017-02-28:
We don't do special cases. A mixer line is a mixer line and always has the same settings regardless of where it is. Especially since they can be moved, and for example shifting a line to another channel would screw it up when it went over the first slot of a channel.

Miami-Mike, 2017-02-28:
The worst part of the problem is that the Multiplex setting isn't visible in Companion on the Edit screen, or Print Model, or Compare Models. In all of those cases the first line lacks a ``+=``, ``*=``, or ``:=``, which misleads the user into assuming that there's no need to display the Multiplex setting of the first mix line since it's automatically set to ADD or the equivalent REPLACE.

The best solution would be to automatically change the setting. The second-best solution would be to display the setting instead of hiding it so that the user at least has a chance to see why his mix doesn't work. But you can't be serious about not wanting to fix this.

kilrah, 2017-02-28:
You can't be serious about your "best solution" being totally inappropriate even after that was already explained to you?!

Displaying the multiplex in all cases could be a valid enhancement indeed, but... that's not what you asked for nor explained initially.

Miami-Mike, 2017-02-28:
If you think your explanation is valid then you should tell the programmer who decided to hide the multiplex setting of the first mix line. It's clear to me that it was deliberately hidden and for a good reason, but unfortunately it seems to have been based upon a mistaken assumption.

The reason the original programmer probably had was that there's no need to know the setting of the first mix because it can be assumed to be such that the mix will have a reason to be there. That's pretty much a no-brainer. What good is a mix that multiplies itself by zero?

The mistaken assumption was that other programmers would see it the same way and not allow useless mixes to exist and confound people trying to learn OpenTX.

But if you want to call your half-baked bug fix an "enhancement" then go ahead. Exposing the bug isn't as good as fixing it but it's better than keeping it hidden.

mpjf01, 2017-03-01:
The empty value has always effectively signified, to me at least, that the field has no meaning. If the multiplex was displayed users might, probably rightfully, assume that it was somehow active - not realising always that this doesn't make sense. Whether it is automatically converted to ADD, or, as it is currently, effectively automatically converted to "no effect", the end result is the same - the field does nothing. I'd prefer to see it left empty as it makes that clearer.

projectkk2glider, 2017-03-01:
Please calm down.

After quick inspection the mixer code does nothing special if the first mix on a mixer channel is MULTIPLY which means that 0 gets multiplied giving 0 as a result. So in effect such line does nothing. 

Now this is not something new, it has been like that long time, the question here is what to do about it, options are:
 * nothing
 * change the display (show symbol if it is not ADD)
 * change the mixer behavior and convert such line to ADD - (I think this is the worst solution)

kilrah, 2017-03-01:
It looks like the symbol was shown in 2.0, and that was lost in 2.1 most likely by mistake when the screen was reworked.

@Miami-Mike you're simply the first to notice / complain about it in 1.5+ year. 

As explained automatically changing multiplex when the line is in first position is NOT acceptable since it WILL break setup when lines are moved. People certainly won't see the symbol disappear when they click the move up/down buttons nor think this could happen.

projectkk2glider, 2017-03-01:
So, we are in agreement: we will fix the the display on the radio. Need to check if Companion has the same problem (with displaying).

kilrah, 2017-03-01:
The whole discussion is about companion editor. 
The first line has never had its mode displayed on the radio. We can try to, but unlike companion screen real estate is limited.

Miami-Mike, 2017-03-01:
Okay, one more try and then I give up.

You guys are objecting to my first post, where I said that MULTIPLY should automatically be changed to ADD. It seems the idea of automatically changing a setting triggered a negative reaction, so I'll concede that and suggest another way to fix this that makes a lot more sense than the solutions you guys are contemplating, which sound like they'll only get you into a deeper mess.

Consider Include Trim, another mix setting. You can choose "No" or leave it at the default "Yes", but "Yes" doesn't really mean "Yes", it means "Yes, **if applicable**". That is, yes if there's a trim to include.

Why not follow that same OpenTX philosophy with the Multiplex setting? When a mix line is set to ADD, MULTIPLY, or REPLACE, by all means leave the setting as it is, but like Include Trim, make it apply only **if applicable**. Add the mix **if there's something above to add it to**. Multiply the mix **if there's something above to multiply it by**. Replace another mix with this mix **if there's a mix above to replace**. Otherwise just treat the mix as a starting value that can be modified by any mixes that follow.

kilrah, 2017-03-01:
This is getting worse and worse...

> Add the mix if there's something above to add it to. Multiply the mix if there's something above to multiply it by. Replace another mix with this mix if there's a mix above to replace

So now you can have a line that has been clearly set to MULTIPLY, and that when you look in your settings everything suggests it MULTIPLIES, but it doesn't actually do that, it ADDs instead.
Then you move it and it it doesn't ADD anymore, it MULTIPLIES now, even when you didn't change a thing.

How can you think anyone will understand that??

jtaylor2, 2017-03-01:
That last suggestion makes no sense.  First off I don't want the program to decide I don't know what I'm doing and try to correct it.  And how can it know that I am done editing (moving things around).  I copy and paste a multiply into the first mix line with the intent of doing an insert  before with another mix.  The program doesn't know what I intend and shouldn't try to guess when I'm done and that I've made a mistake.  If I've made a mistake my testing will show it.  Adding the indicator back like it was in 2.0 would be nice, but nobody (except you) has noticed in several years.

Miami-Mike, 2017-03-02:
Here's my question to kilrah:

> So now you can have a line that has been clearly set to Include Trim: Yes, and when you look in your settings everything suggests it includes trim, but it doesn't actually do that, instead it doesn't include any trim at all. Then you change the source and it doesn't fail to include trim anymore, it includes trim now.

>How can you think anyone will understand that and not just end completely confused?

The comparison is fair; what's good for the Multiplex setting is good for the Include Trim setting. You guys can't have it both ways. If you really believe your own arguments then you need to get to work on the Include Trim setting.

kilrah, 2017-03-02:
> The comparison is fair; what's good for the Multiplex setting is good for the Include Trim setting

Indeed. 
Which is precisely why both have the most generic choice by default. Trim is by default the one of the selected source if any, which is the most likely and "natural choice" until you go change it if you need. Multiplex is by default Add which is the most likely and "natural choice" until you go change it if you need.

Miami-Mike, 2017-03-02:
You're calling your subjective preferences "generic" and "the most likely and natural choice", and using that as an excuse to violate your own opening argument that "we don't do special cases". If you're going that way instead of following a consistent, well-structured, and logical paradigm then this argument is hopeless. You can't reason a man out of a position that he didn't reach through reason.

kilrah, 2017-03-02:
Your preferences are just as subjective, and we can disagree with them in the exact same way you disagree with ours. Your choice not to use OpenTX or to make your own project if you're not happy.

What matters most in this case is that there is logic that WE have chosen for OUR project several years ago, we still agree to that logic, and 100k+ users have been using it for all those years without complaining so it must not be that "wrong" or excessively confusing. 

We're not going to force a change that isn't even logical to us to all our users for one guy who probably got stuck for a couple of hours once on a wrongly set multiplex and can't get over it. We'll make that mistake less likely by making the setting more visible, but not change the behavior.

Miami-Mike, 2017-03-02:
> There's also an entire background of technical reasons and limitations that needed to be taken into account and have contributed to directing things towards where they are even when that meant deviating from the "ideal" solution.

Okay, you don't want to fix it because it's too hard. Now that you've admitted that it puts this whole discussion in a different light and makes me feel guilty about asking too much of volunteers.

So I'm sorry I've been bothering you guys. I guess I'll just keep quiet and find my own workarounds, like I did with the 250/1024 =  24.4140625% trim bug.

kilrah, 2017-03-02:
Frankly at this point your atrocious condescending and accusing attitude gives us exactly 0 desire to listen to you regardless of how relevant your remarks are. We've been able to do well for 5 years without you, we can continue to do so without you to telling us what we should do. 

Please refrain from further "contributing" until you cut down on the attitude.

As a reminder, your first post in this thread wasn't "hey guys, there's something that may be inconsistent or confusing, what would you think about changing it to X becasue Y", it was "has this been changed to how I want it to be yet?" with absolutely no desire from you to listen to our arguments.

Miami-Mike, 2017-03-02:
> Frankly at this point your atrocious condescending and accusing attitude gives us exactly 0 desire to listen to you regardless of how relevant your remarks are.

So your willingness to fix a bug is dependent upon how you learned about it. If you don't like the person who made you aware of it then you won't fix it.

Well, that's sure to give everyone great confidence in the integrity of OpenTx.

kilrah, 2017-03-02:
No, but since it wasn't a bug... it was at most a change request that was presented as "you will do that now becasue I say so".

commit 76f5d029 (2017-03-22): Warning when Multiply used on first mix of a channel, fix #4525 (#4647)

* Proposal to fix #4525

* Clearer text

* Companion side

* Translation not needed
closed 2017-03-22 by kilrah