opentx/opentx #4526: Rssi warning read from the Horus is wrong in companion

RealTadango, 2017-02-28:
Companion reads 41 for warning but the value in the Horus is 45. Critical value is correct.

RealTadango, 2017-02-28:
It was for all models.... a second read from the same sd contents did show the right value strangely?

mpaperno, 2017-03-03:
Steps to recreate?  Going between simulator and Companion seems fine here.  Is it consistently a problem on "first read" (whatever that means) but not subsequent "reads?"

If this is a self-built version  you're testing, please realize there are dozens of build option combinations and other variables (compiler, OS, Qt, etc...) which may affect the outcome.  When we ask for details about the build, it is for good reason(s), and we need full information.  Thanks.

closed 2017-04-10 by mpaperno