opentx/opentx #4527: Companion crashes after simulating model with dangerous module functions

RealTadango, 2017-02-28:
When using dangerous module functions and simulating the model from companion companion crashes when closing the simu except when you browse specal functions settings on the simu.

Companion also does not show the special functions correctly....

mpaperno, 2017-02-28:
Could you please post a step-by-step to reproduce the crash?  I don't see it here, but quite likely I don't understand what you mean exactly.   Any particular radio type?

I do see that the "dangerous functions" are not in the list of available functions in Companion.

RealTadango, 2017-03-01:
Steps to reproduce
 - Build firmware with special functions and run it on the Horus
 - Add a special function for Rangecheck for a model
 - Import model into Companion
 - Simulate the model, don't do anything in the simulator
 - Close the simulator and crash

If you visit the special functions screen in the simu it won't crash. Companion always shows Override channel function and not the Rangecheck

The simulator does show Rangecheck when i check the special functions but i cannot select Rangecheck myself. Option list is limited but the text resources are working :) I did not build the simu with dangerous model functions

3djc, 2017-03-01:
The simu uses a 'fairly standard' users configuration, and until that changes "at some point in time", it cannot be expected that eeprom saves made by different firmware settings will not produce errors or unexpected results

RealTadango, 2017-03-01:
but the values are correct (although not displayed correct). The main issue is the crashing of companion. Just add a try / catch and ignore the simu values that do not fit in companion.

kilrah, 2017-03-01:
>  I did not build the simu with dangerous model functions

Ah that is most likely the problem.

RealTadango, 2017-03-01:
No, i build the simu with dangerous module functions and it crashes even more :) When i select a new special function with range test and exit the simu companion crashes also....

mpaperno, 2017-03-01:
I don't have a Horus.  If I follow the repro steps purely using the simulator, I don't get a crash.

Here is how the special function is set up... is that right?
[image]

It even seems to "work"  :)
[image]

Could you post your .otx file which causes the crash?  Thanks!

mpaperno, 2017-03-01:
Also, which operating system, compiler, and Qt versions?

mpaperno, 2017-03-01:
>Companion also does not show the special functions correctly

As for this, Companion right now looks for `danger` flag in the firmware ID string before displaying these functions in the list.  But there's no way I can see to select that as a Build Option in the radio profile.  Perhaps this is by design (on purpose), but I don't know.

schwabe, 2017-03-01:
Yeah that is by design. And after dealing with users for a bit I now know why.

RealTadango, 2017-03-01:
See attached .otx. model CT-114 Tutor has the issue.
[28-02-2017.otx.txt](https://github.com/opentx/opentx/files/811036/28-02-2017.otx.txt)

RealTadango, 2017-03-01:
When i add a SF with Rangecheck on the simu it is not stored, or not transferred to the .otx, so no error.

mpaperno, 2017-03-01:
I still get no crash with your .otx.  Are you sure you have `DANGEROUS_MODULE_FUNCTIONS` defined in your `libsimulator` build?

> When i add a SF with Rangecheck on the simu it is not stored, or not transferred to the .otx, so no error.

`simu` or `simulator[22]`?   (We still haven't established the OS here.)

You can start the standalone Simulator with a .otx file.  From there all changes are saved back to the .otx file on exit.  When Simulator is launched from within Companion, changes are not saved in any fashion.

RealTadango, 2017-03-01:
I build sim with the right options because it showd them in the dropdown. I build for windows, on windows ;)

mpaperno, 2017-03-01:
What dropdown?  😕

RealTadango, 2017-03-01:
Special function sector...... i do to much html programming :(

mpaperno, 2017-03-01:
Then I don't know.  W/out being able to reproduce, impossible to fix.  I've tried Qt 5.7.1 with both MSVC2015 and MinGW-w64. 

If you know how to run a debug session, that would be my next suggestion.

Barring that, you should try generating the project config with `WIN_USE_CONSOLE` CMake option enabled (or `-DWIN_USE_CONSOLE_STDIO` added to compiler definitions for `companion` target).  Then launch companion.exe from a command line.  This will allow you to see debug output and possibly some clues before the crash happens.

closed 2017-04-10 by mpaperno