opentx/opentx #4529: 2.2 Companion Simulator intermittantly only partially works

mythreeleggedcat, 2017-03-01:
2.2 RC10  on Windows 7 SP1  
May be duplicate with #4224 but that issue seems too overloaded and this behavior seems a bit more specific.

Running Companion Simulator for the Q-X7 or X9D+  from the Model editor typically every other time but sometimes more or less frequently, the Simulator only partially works.  Specifically, the main control display animates as expected but this is not reflected in the monitors in the tabs.  The Output Tab does not reflect any of the stick moments.   With the Q-X7 (haven't checked the X9D+) the radio's model monitor display does not reflect the stick movement either.  I think I saw that the GVARs don't update although I'd have to check again to be sure.

The problem does not seem to happen (or happens much less frequently - I don't recall is failing) if the Simulator is started from Simulate TX buttons on the .otx screen rather than the model editor

powercycle, 2017-03-01:
I can confirm this issue on Linux (Ubuntu 16.04).  However, since the initial few times that it happened to me, I can no longer make it occur using the X9D+.   I didn't have the X7 firmware built so couldn't validate that.

mpaperno, 2017-03-01:
This is most likely fixed by #4342 (and possible duplicate of #4346).  Please re-open if still happening in next RC (11).

mythreeleggedcat, 2017-03-07:
This issue still occurs in RC11 with about the same frequency.  So far I have only tried the Q-X7 simulator in the new version.

I installed over the top or RC10 without explicitly uninstalling in case that is relevant.  Companion reports that it is version 2.2.0N363 and the simulator is clearly the new version with side icons and separate monitor window

3djc, 2017-03-07:
Windows, Linux, MacOsx ?

mythreeleggedcat, 2017-03-07:
hmm thought I typed it in that comment...but in any case Windows 7 SP1

3djc, 2017-03-07:
Yes yes, I saw that too late sorry. Does it happens with a specific otx loaded, or even when starting from empty ?

mythreeleggedcat, 2017-03-07:
I created a new one to try something.  I haven't tried an existing file.

mythreeleggedcat, 2017-03-07:
I just tried a converted .X9D+ .otx file and the problem reproduced there too.  Happened about 1 in 5 tries when using simulate from the model editor.  The monitor window showed no stick movement.

The .otx was converted from an OpenTX 2.0 .eepe

Think it was more frequent with the Q-X7 model simulator.

As before it seems the problem is less likely to occur if you use "Simulate TX" from the Model List.

mpaperno, 2017-03-07:
Really need a reproducible case for this...  I run Win 7SP1 primarily and have launched Simulator hundreds of times in various scenarios and have not once seen anything like this.

Could you tell us more about the computer specs?  Do you have another machine to try it on?  Maybe post your .otx... I doubt it has anything to do with it, but who knows.

I'm also just now re-writing how the simulated radio inputs are handled, so this may possibly help things (it will not be as timing dependent).  Perhaps I'll post a test version here, see if it makes any difference....

projectkk2glider, 2017-03-07:
@mpaperno you should try with `-DCMAKE_BUILD_TYPE=""` or even `-DCMAKE_BUILD_TYPE="Release"` to see if you can reproduce it then.

mpaperno, 2017-03-07:
I have it both ways (and some "Debug" also), across platforms and Qt versions and toolchains (lots of combos possible!).  I guess more to the point would be trying the official RC11 version...

mythreeleggedcat, 2017-03-07:
My machine is an older HP with an AMD Athlon 64 Dual Core 4400+ 2.30Ghz running Windows 7 Professional with SP1.... with with stock no pen or touch screen, 6G RAM,  integrated geforce 6150SE graphics and a 256G SSD drive
 (ie nothing fancy or peculiar)  except maybe SSD.

I very much doubt it has anything to do with OS or hardware as such.  Feels like a timing issue with the initialization of the monitor window to me.

I can try it on a different machine later - maybe tonight - if you think it will inform - I have a slower one and a faster one  - all windows 7, I think...but I suspect all that will say is intermittent timing problems are hard to reproduce.

mpaperno, 2017-03-07:
Hold on... we're talking ONLY about the Radio Outputs window/widget?  The actual controls all work correctly otherwise?

[image]

mythreeleggedcat, 2017-03-08:
yes...the main window appears to respond correctly.  It's the monitor window, , aka "Radio Outputs"  that freezes.  That was the original issue I reported.   Sorry if that wasn't clear.

mpaperno, 2017-03-08:
OK, sorry, I misunderstood it to mean that none of the mixers actually worked (which is what #4342 was about).  But if the channels monitor on the simulated radio itself works (assuming that is what what you mean by "appears to respond correctly"), then this is an entirely different issue.  I will take a closer look at the Radio Outputs window specifically then.  Thanks!

If anyone else can reproduce this I'd appreciate knowing the circumstances.

elecpower, 2017-03-08:
Struck it once a week or so ago and in my case I suspect it was that there were radio startup warnings like switch positions. Clearing those fixed the problem but it might just have well been a focus issue. With so much activity didn't bug as it went away. Anyway something to consider.

mpaperno, 2017-03-08:
There are no outputs until all warnings are cleared, that is expected.  This is in the radio code itself, by design (as far as I know).

mythreeleggedcat, 2017-03-08:
No sorry I misunderstood you and did not check.  

When this occurs, the radio simulator channel monitor does not respond  **and**  the Radio Outputs windows appears but nothing animates on it.

The radio simulator basically works...you can navigate and set things but the channel bars don't move in the monitor page.  The battery indicator flashes there and the trim sliders move the "thumbs".  The trim buttons do not beep or move the thumbs though and "trim center" does not play.  

When it works the trim buttons work, beep and  "trim center" plays

3djc, 2017-03-08:
Yes, no outputs until warnings are cleared is by design (on the radio that's no  power to the modules)

mythreeleggedcat, 2017-03-08:
As I noted, this happens when you run the simulator from the model editor.  When run like there,  the start up throttle and switch checks are never run...or at least, never displayed.

You only see the startup checks if you run the simulator from the Model List window...which is maybe why it doesn't seem to fail when run from there.  At least I don't recall seeing it fail, running from there.

mpaperno, 2017-03-12:
@mythreeleggedcat I have a version for you to try, if you could.  

The zip file contains just the two .exe files (companion and simulator) and two simulator libraries, for the X9D+ and X7 (the ones you mentioned originally).

To use, you basically just need to drop them into your existing RC11 install folder.  You could backup/rename the 4 files which will be replaced as a precaution (or just re-install from RC11 download to go back).   If you need more help with any of that, let me know.  

It's possible the old simulator libraries (for the other radios) will not run with this version of Companion/Simulator.  I haven't tested.  This version also includes all the other changes so far since RC11.

Thanks!

[companion-rc11-timer_test-12-03-17-01.zip](https://github.com/opentx/opentx/files/836388/companion-rc11-timer_test-12-03-17-01.zip)

mpaperno, 2017-03-12:
First version had a mistake (a bit surprised it worked at all).  Updated file in previous post.

mythreeleggedcat, 2017-03-13:
Unfortunately I  just updated to RC12...also installed over the top or RC11.  The problem is still reproducible but seems much harder to reproduce...fails say 1 in 10 times rather than works 1 in 6.

Do you still want me to try the above?  I can try re-installing RC11 but probably that only makes sense as a fresh install.

mpaperno, 2017-03-13:
Yes please test it out.  Even better to compare with RC12 anyway.  You can still simply drop the files from the zip archive I posted into your RC12 install folder, no problem.  Just backup the existing files if you wish, as I mentioned before.  You could also make a copy of the whole Companion install folder and make changes in the copy (simply dbl-click the companion.exe file directly to run it vs. using the installed shortcuts which point to the original copy).   Thanks.

mythreeleggedcat, 2017-03-13:
ok...will see if I can get to it tonight.

mythreeleggedcat, 2017-03-14:
Same behavior... the channels indicators intermittently do not animate in the Radio Outputs window or the Channel monitor in the simulated radio interface.  Its seemed easier to create they with the pure RC12 install but that could be luck.  I'll try some more tomorrow.

Note: I tried it by copying the ProgramFiles(x86)\OpenTX\Companion 2.2 folder and copy the 4 files over.  I binary compared them to make sure they got copied.  I ran companion.exe directly from the copied folder

mpaperno, 2017-03-14:
Thanks for trying.  Well it was a shot in the dark, admittedly.  Otherwise I may need to resurrect my old Core Duo machine to give it a try  :) (Your machine is fine it's just that I think most/all of the devs are using newer/faster CPUs so we're not getting much testing done on older ones.)  

I do have a whole new version on tap which changes a lot of the timing interactions between the Simulator UI and the radio firmware, perhaps that will help.  That will most likely be for next OTx, 2.2.1 or whatever.  But if you're up for more testing, I could post that version for you.

mythreeleggedcat, 2017-03-14:
No worries on insulting my machine :) Let me know if you want me to try something.

mythreeleggedcat, 2017-04-05:
Just FYI - this still occurs in RC14 2.2.0N366

mpaperno, 2017-04-10:
Please let us know how it goes with the latest RC15.  Major changes under the hood in Simulator since RC14 in terms of performance and synchronization/timing issues.

mythreeleggedcat, 2017-04-13:
No apparent change with RC15.  As before, using a simple default model created by cancelling the wizard.  Seems to work if I simulate the radio from the model list...Works intermittently when I simulate the model from the model editor.  

The problem as before is that the stick movement and channel monitor graphics on both the simulator and Radio Outputs are frozen.  Interestingly when the problem occurs, the trim sliders still animate on both the control panel and the simulation but the trim buttons only animate on the control panel but do not move the sliders on simulation and there is no beep.

mpaperno, 2017-04-16:
I'm baffled here, especially after all the internal changes since this was first reported.  At this point I don't think it's a timing issue.   It's also entirely possible I don't really understand what is going on (the symptoms) when the problem happens.  Especially since we had the confusion earlier.   It sounds a lot like what happens when there are warnings on the radio screen (all outputs + trims are dead until warnings cleared).

@mythreeleggedcat Is there any way you could record what is happening?  A short video might clear things up (or not).   Also do you have a different computer to try with?  That would go a long way to narrowing down the cause, I think.  Thanks.

mythreeleggedcat, 2017-05-12:
Sorry wasn't paying attention and just saw this.  Problem still occurs in RC18 N371.  I will attempt to record something.  I have a couple of other computer I can try.