opentx/opentx #4533: Taranis Q X7  Telemetry gps display issue

wolflonezx11, 2017-03-02:
hi all 
opentx rc10

1. Taranis Q X7  Telemetry GPS display issue

00.23
21.45  

Display issue

2. Taranis Q X7  Telemetry  VOTG MENU Display 불량 
menu Select none

3djc, 2017-03-02:
1. fixed already, will be available in next RC

2: no idea what you are talking about

Jmelz, 2017-06-23:
On Q X7, telemetry pages only show GPS with 2 decimal places which leads to a very big area of uncertainty (inaccurate):

[image]

3djc, 2017-06-23:
With some coordinates, all the available space is fully utilized. If you want to allocate more space to it, you can create an Lua display for it

Jmelz, 2017-06-23:
Sounds great, but I'm a noob :)

It seemed like there was enough room for 2 more digits, which is why I came here to ask.

3djc, 2017-06-23:
Whats your GPS source ? On mine : 

[image]

Jmelz, 2017-06-23:
I am speaking of NMEA, not Decimal Minutes Seconds. I can see your point about degrees minutes seconds filling the screen, but NMEA does not.

The source is a bit complicated, but I know it feeds okay into Tower on Android. Eagletree Vector --> Dragonlink v3 receiver --> Dragonlink v3 transmitter --> PPM input on back of Q X7.

3djc, 2017-06-23:
You need to account for 1 more digit upfront since it can have 3 digits, I think it won't leave two more spaces

Jmelz, 2017-06-23:
Not sure I follow you :(

If you change to NMEA in main settings, you'll see. There is space for more digits, it's just a design decision.

3djc, 2017-06-23:
It also matches x9 display in NMEA:

[image]

3djc, 2017-06-23:
What I mean is your lat starts with 2 digits, it seems like a lot of room left, but not for this guy in Asia for exemple:

[image]

Jmelz, 2017-06-23:
Okay, understood - but when the person in Asia uses the other format, does it not also push over?

[image]

3djc, 2017-06-24:
Then just launch the LUA when in that situation (by the way, in 2.2, you'll get last transmited coord even if your RF link if down ;))

[GPS_Coord.zip](https://github.com/opentx/opentx/files/1099475/GPS_Coord.zip)

[image]

Jmelz, 2017-06-24:
Thanks very much! I was told that the last coordinates would be logged, but so far I'd not had any luck seeing the coordinates on the screen in any scripts after powering down the plane (killing the link). I'm running 2.2 already.

3djc, 2017-06-24:
You can have an updated version of the script running in the background too, to record the value while available. Have a look in the opentx Lua doc

Jmelz, 2017-06-24:
Yes, I noticed that. I also saw the previous author used it in the script I am modifying:

return{init=init,run=run,background=background}

but even in his script, it seems to blank the coordinates when telemetry is lost... Must be something I'm doing wrong or not understanding.

closed 2017-11-10 by 3djc