opentx/opentx #4539: 2.2 RC10 Mix Scripts not accepting 8 inputs

RCdiy, 2017-03-02:
Mix Scripts, input table, accept 8 inputs under 2.1.9 but under 2.2 RC10 the following occurs
- CUSTOM SCRIPTS configuration screen only the first 6 inputs display
- Script error when referencing an input above the 6th position on the input table
Sample script attached...

[mxtst.lua.txt](https://github.com/opentx/opentx/files/814531/mxtst.lua.txt)

3djc, 2017-03-02:
This is a result of this change : https://github.com/opentx/opentx/commit/afce17443fc5243c0e944a9abef73bfe63a9

@bsongis or @projectkk2glider can you confirm this is ok to put MAX_SCRIPT_INPUTS back to 8 ? (meaning relevant eeprom changes have been done)

bsongis, 2017-03-02:
It has been done on purpose, 8bits were not enough to store all the sources we have now and we have moved to 16bits. Therefore we reduced the number of inputs.

bsongis, 2017-03-02:
Even if we decide to return to 8 inputs at the end, we cannot change this in 2.2, it means an EEPROM change

3djc, 2017-03-02:
Ok, will update the LUA doc accordingly

RCdiy, 2017-03-02:
Documentation needs two updates. One under input table documentation and one under changes between/migration 2.1 to 2.2.

3djc, 2017-03-06:
Doc updated

closed 2017-03-06 by 3djc