opentx/opentx #4542: X7 wil restore model backups from X9, causing a messy and unusable model setup.

lshems, 2017-03-02:
Apparently there is no check on a restore on X7 from backups of X9. At least confusing?

3djc, 2017-03-02:
Nor is there check between x9, but that can't happen by accident, since you need to swap sdcard. A possible future enhancement I guess

lshems, 2017-03-03:
People are going to backup their models, exchange them, and transfer them from X9 to X7 and vice versa.

This could be dangerous.

Off course, I know, check the settings, check the outputs in the monitor, check everything before switching on the model.

But everybody knows this is only a liablity scheme. People won't do that all, and things will happen, and thus is a causing a risk.

Just my cup of tea, do what you want.  
It's not your responsibility !! ?? according to the software license.

3djc, 2017-03-03:
Clearly, transferring models between different radio is not a trivial task, and can rate from quite easy to completely impossible in some cases (because for example my x9e has 16 switches, there is absolutely no way my backup can be transferred to even a standard x9e).

I'm unsure what it the best response to this scenario : block same physical radio ? same type ? same version ? Not a trivial issue by any means

kilrah, 2017-03-03:
As mentioned on #4512 handling those cases would be beyond practical and maintainable when even possible at all.  

Allow only same radio/version is the only way... and isn't even foolproof. Ideally the whole function should be dumped, and replaced by the possibility to store additional models on the SD card Horus-style.

lshems, 2017-03-03:
Well, there is a difference between foolproof and inituitive behaviour. Intuitive would mean for me that a check on model and version would be neccesary, if they are not automatically converted correctly.

In the field it is quite normal to switch sdacrds to load models backuped by another radio.

It's not for me, I know what I am doing. It's for others, and up to you to decide.

kilrah, 2017-03-03:
After a quick check we can't even do that without major changes - Radio type and version are only stored in the radio settings, there is no such information in the model so we can't even know if it's compatible or not.

lshems, 2017-03-03:
OK, so some care has to be taken then. 

Fine with me, glad I know.

Thanks

3djc, 2017-03-04:
[image]

You actually cannot import X9 from X7 already (that check was added a while ago)

closed 2017-03-04 by 3djc