opentx/opentx #4546: Telemetry for the new Scorpion ESC Tribunus

Storchy1, 2017-03-03:
Is it possible to integrate the new Scorpion ESC Tribunus Telemetry?
I have asked Scorpion for the Telemetry Data Sheet and i got it.
Thanks a lot
[Tribunus ESC protocol description.pdf](https://github.com/opentx/opentx/files/815916/Tribunus.ESC.protocol.description.pdf)

3djc, 2017-03-03:
Well, you can't simply connect your esc to the receiver smartport. If you want your ESC telemetry, an interface (I mean a real physical board with a small cpu) will need to be build to connected to you 4,6,.. ESC on your craft, speaking their language, and to a single smartport to receiver speaking FRSky language.  Such an interface to a single ESC can probably be done at the flight controller level, but their developer would be better suited to respond to that question There is not much we can do at our end to replace that.

Storchy1, 2017-03-03:
Thank you very much, 
i think i have found the little interface you are talking about.
[http://j-log.eu/s32/s32-de/scorpion-tribunus/](url)
This interface can understand Tribunus and can translate to Frsky SmartPort.
This can be closed

closed 2017-03-03 by Storchy1