opentx/opentx #4547: Spacing issue for DBLSIZE number on 128x64

floaledm, 2017-03-03:
Kind of a followup on #4540 

Looks like lcdDrawNumber() in lcd.cpp for the 128x64 display hasn't been updated to match the updates made for the 212x64 display (see #4157).

This results in display inconsistencies (shown using Companion 2.2RC10):
[image]
[image]

The code I used to display this is:
`lcd.drawLine(0, 0, 0, 16, SOLID, 0)`
`lcd.drawNumber(32, 0, 332, DBLSIZE+PREC1+RIGHT)`
`lcd.drawLine(32, 0, 32, 16, SOLID, 0)`
`lcd.drawNumber(61, 0, 333, DBLSIZE+RIGHT)`
`lcd.drawLine(61, 0, 61, 16, SOLID, 0)`
`lcd.drawTimer(63, 0, model.getTimer(1).value, DBLSIZE)`
`lcd.drawLine(105, 0, 105, 16, SOLID, 0)`

3djc, 2017-03-03:
It is already fixed for RC11

closed 2017-03-03 by 3djc

floaledm, 2017-03-03:
Thanks! Can I get the reference to the PR to take a look on how that was done?

3djc, 2017-03-03:
This PR is scheduled to go live before rc11, we are just debating some details about it

floaledm, 2017-03-03:
Sounds good... I wasn't finding the changes on the branch so I was wondering where they were... Many thanks!