opentx/opentx #4553: Flickering Logical Switches

1Sven, 2017-03-04:
If you set a logical switch in Mixerscript, the switches will not stable.
Create a new model. Make a *.lua like Test.
Set a SF(61) for playing a track like “3s” 1X   and set the slider that you can hear the voice.
Now you can hear the sound many times randomly. In real Horus it is more than in simulator.

[Test.zip](https://github.com/opentx/opentx/files/818853/Test.zip)

kilrah, 2017-03-04:
This is not "setting" a switch i.e. setting its state, it's "reconfiguring" the switch dozens of times per second. You should never be doing that.

closed 2017-03-04 by kilrah

1Sven, 2017-03-04:
I almost thought of that...
I can put as many switches as desired. I use this for the flight modes. 
The script I always use looks like in the "JoXY".
Is there a different solution without many logic switches in the standard interface to put?
With the solution of the script you can make a lot of switches on a slider, wery usefull for funktionelly modells...
[JoXY.zip](https://github.com/opentx/opentx/files/818945/JoXY.zip)

kilrah, 2017-03-04:
I might be wrong, but as far as I know that's totally inappropriate as well.
The correct way using Lua would be to output the X and Y values to Lua outputs, then assign those outputs to free channels in the mixer, then test the channels in the logical switches. 

Which means... a Lua script is useless since you can test the X and Y sources directly in the LSs.

If you really want to use a Lua then I believe a valid workaround would be to keep track of the state of things inside the script, and ONLY reconfigure a switch if it has changed, not everytime the run() function is executed.

@projectkk2glider am I on the right track?

1Sven, 2017-03-04:
Ok I understand. But since 2.2 I have used that. The servos are too slow to follow the loop. Let's see what I'm doing. It was very convenient. Thank you!