opentx/opentx #4557: RC11 N363  X9D+ At Simulator No Mixermonitor only Channelmonitor

hrenz, 2017-03-05:
Hy,
Profil X9D+

No Mixermonitor at Simulator with ENTER
At Option Mixermonitor is selceted

( not tested at X9E)

kilrah, 2017-03-05:
Simulators don't use options.

hrenz, 2017-03-05:
Mixermonitor at simulator is importend 
 for programming and testing before got to Tx an Modell ans set Servo directions

kilrah, 2017-03-05:
We'll spend time figuring out what options to use by default but that will be only in final release, not RCs.

hrenz, 2017-03-05:
ok I see

mpaperno, 2017-03-06:
I think there was actually consensus for removing the option altogether and just having the MM always available in every build, but the changes slipped for RC11.   (Literally the only difference at this point is if the "enter" key/action in channels monitor has any effect or not (there is no extra memory or resources used, etc.).  Seems like even if someone DID get confused by that, it should be fairly obvious that repeating the key action gets you back to the default channels view.)

kilrah, 2017-03-06:
Ah yes, forgot about it

mpaperno, 2017-03-06:
Me too :)   I'll reopen this issue to remind us.

PR #4573 (2017-03-11): Remove mixermon option (make it always true)
This fixes #4557
closed 2017-03-11 by bsongis