opentx/opentx #4558: Memory consumption while changing models?

pau33, 2017-03-05:
I use an lua mix script as a server and a telemetry script for showing the values (global variables defined in the mix script).

As we found out in an earlier issue (#4162) my mix script is at its memory boarder. Fortunately I don't have to change much there.

A good indication for me, that memory in the mix script is getting low, is when the telemetry shows up an error in an arithmetic expression regarding one of the global variables of the mix scrip.
(Taranis X9D Plus OTX 2.1.9)

I now have the following behavior. I can handle it, but it might worth noticing:
I have different model versions and even different models using the same mix and telemetry script (sometimes with slightly different parameters which are adjusted in the startup procedure of the mix script).
When I now change between the models, I get the above error in the telemetry script indicating memory problems in the mix script. When I then turn the radio off and on again, everything works fine. When I change to a model without lua scripts an back again also everything is ok.

For me looking at a black/gray box it seems as if there is some memory which might not be cleared between switching models that leads to this behavior. Or the reload procedure ist somehow not fully correct.

Is this a general issue?

lshems, 2017-03-06:
Yes, in that way that I noticed also some unpredictable behaviour around memory errors. When loading some code, and unloading it again, which should return the LUA environment exactly in the same initial state, and even the memory reported by the garbage collector is the same, several cycles will eventually crash the LUA environment.

LUA is quite peculiar in memory management, but I to have the impression there is some memory problem. But.... where to start looking for it??

I've no idea.

projectkk2glider, 2017-03-06:
You must both remember that memory is managed by memory allocator (included in library) which uses unknown algorithms to handle out the memory chunks. When some of the used chunks are freed and some stay allocated one always gets some amount of free space fragmentation which means that any subsequent run of the same script experiences different memory environment and might fail some allocation (if the script was close to the memory limit). 

Then again it might be some other bug...

mpaperno, 2017-03-06:
@lshems Could you post a script which causes the crashing (and steps to reproduce)?  I presume this only happens on the actual radio? 

Though Damjan is correct, fragmentation could cause memory "leaks" which the Lua memory counter is most likely not aware of.  I believe the default allocator just uses `realloc`.

lshems, 2017-03-07:
I have worked around this issue in several ways, so I cannot reproduce it easily.

I think paul has a script that reproduces this effect. I'll ask him.

pau33, 2017-03-07:
I will be back home at the weekend, then I will look for the script.

lshems, 2017-03-23:
Paul,

could you check if you can find the scripts to reproduce the memory error on multiple loads/unloads of the same code?
Thanks!

projectkk2glider, 2017-03-23:
I have a new idea: trace all Lua (de)allocatons together with the source and line number where they happen. This could then be graphed or parsed somehow to determine where the problem is and how Lua memory is being used. Stay tuned.

pau33, 2017-03-23:
I think I have a configuration and lua files where we can reproduce the situation. Is there a private channel where I can share the files?

projectkk2glider, 2017-03-24:
You can email them to me, my emails is in my profile.

Meanwhile example of lua memory usage gathered wit the tracer:
[image]

lshems, 2017-03-24:
Will this tracer be available in the companion?

pau33, 2017-03-24:
I am very sorry, I looked for my files before I wanted to send them, but all of a sudden, the error, which I had the last time, disappeared. The only thing I changed in lua, were 3 simple number-entries in a field. That's nasty.
I can send you the files if you want or I wait until I see the error again?

lshems, 2017-03-25:
Did you upgrade you opentx version?

projectkk2glider, 2017-03-25:
@pau33 without the reproducible case files are of no use to me.

@lshems not sure yet, but the plan is for the tracer will be available both on the radio and in Companion.

lshems, 2017-03-25:
Got some buffer errors. What I do is simply compiling a lot of script files. Looks like the same type of problem: recurrent loading and unloading of functions. 

I'll try to make an example reproducing this error:

f_open(C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.lua, 1) = 0CFCCF30 (FIL 11E9F1AC)
f_close(0CFCCF30) (FIL:11E9F1AC)
convertToSimuPath(): /SCRIPTS/WIZARD/wizard.luac -> C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac
findTrueFileName(C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac)
	found in map: C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac
f_open(C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac, a) = 0CFCCF30 (FIL 11E9F3B0)
f_close(0CFCCF30) (FIL:11E9F3B0)
convertToSimuPath(): /SCRIPTS/WIZARD/wizard.luac -> C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac
findTrueFileName(C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac)
	found in map: C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac
f_utime(C:/Users/Public/Documents/sdcard-taranis-x9-2.2V0005/SCRIPTS/WIZARD/wizard.luac) set mtime = Mon Mar  6 20:37:40 2017

luaDumpState(/SCRIPTS/WIZARD/wizard.luac): Saved bytecode to file.
f_close(00000000) (FIL:0DB5B978)
GC Use Scripts: 90340 bytes
[W] DebugOutput::onDataBufferOverflow():207 - Data buffer overflow by 2 bytes!
GC Use Scripts: 54407 bytes

lshems, 2017-03-25:
MMM, might be something else. But it is reproducible. I have no clue this is just a warning, or things are really going wrong.

These are the scripts I am using:

/SCRIPTS/telemetry/comp.lua:
`local listfile="/SCRIPTS/list.txt"
local addChar
local file2compile
local list 
local pos=0
local fileNo=0
local dummy

local func=function()
  if not (addChar=="") then
    fileNo=fileNo+1
    file2compile=""
    addChar="/SCRIPTS/"
    list = io.open(listfile, "r")
    io.seek(list,pos)
    while not((addChar=="\n") or (addChar=="")) do
      if not (addChar==" ") then
        file2compile=file2compile .. addChar
      end
      addChar=io.read(list,1)
      pos=pos+1
    end
    io.close(list)
    if addChar=="" then
      return
    end
    --print(fileNo,pos,file2compile)
    file2compile=string.gsub(file2compile,"\\","/")
    dummy=loadScript(file2compile,"tc" )
    print('got to fileno ',fileNo)
  end
--  dummy=loadScript("/SCRIPTS/test.lua")
end

return {run=func}`

and then a /SCRIPTS/list.txt file for the files to compile:
[list.txt](https://github.com/opentx/opentx/files/870400/list.txt)

mpaperno, 2017-03-25:
Are you referring to this buffer overflow: 
`[W] DebugOutput::onDataBufferOverflow():207 - Data buffer overflow by 2 bytes!`

This is from the debug console on the simulator.  Not the firmware.  It's meant to inform you that you may be missing some output in the console.

lshems, 2017-03-25:
Ok, well, that's not a problem then. Thanks for the info.

I've been testing the 2.2n365, on the x7, and got no unexpected out if memories. With the taranis, i need o test. Reason could be that less data ous use by my scripts on the x7, due to smallr graphs on a tiny display. Well see.

closed 2020-04-06 by stale[bot]