opentx/opentx #4562: X7 LCD

Laiwxn, 2017-03-06:
Hi，I want to change my X7's LCD into a large size,but I don't know the Driver IC of the LCD.Is anybody know that?Thanks.

3djc, 2017-03-06:
I'm closing this since it is not at all an opentx bug. If you mean changing the lcd resolution, you have to build yourself a unique version of opentx that works with x9 gui and x7 internals. Not that it can't be done, but it is a fairly significant task.

closed 2017-03-06 by 3djc

Laiwxn, 2017-03-08:
Just change the size not the resolution.So I don't need to rebuild this.

kilrah, 2017-03-08:
I believe it's an ST7565R, but you can probably look in the code to confirm.

Laiwxn, 2017-03-09:
I have checked the code but I just find the initial code.I have not find the Driver IC.

发自网易邮箱大师
On 03/08/2017 21:05, Andre Bernet wrote:

I believe it's an ST7565R, but you can probably look in the code to confirm.

zhaocundang, 2021-01-26:
@Laiwxn 老铁，怎么改LCD屏幕大小