opentx/opentx #4563: OPENTX RC11  XJT  D8 No selection menu

wolflonezx11, 2017-03-06:
HI 

TARANIS Q X7  or TARANIS  X9E 

MODEL SETUP->INTERNAL   RF -> MODE  ( D16 /  L12 ) ?? D8  No selection menu

Can not use D4R-ii.
Please take immediate action
please

schwabe, 2017-03-06:
Are you using a EU firmware?

kilrah, 2017-03-06:
Flash firmware without the EU lock.

closed 2017-03-06 by kilrah

wolflonezx11, 2017-03-06:
non EU firmware

schwabe, 2017-03-06:
as @kilrah said the only firmware we have that do not allow selecting D8 are eu firmwares.

kilrah, 2017-03-06:
You're likely confusing rf firmware version with the openTx eu option.

wolflonezx11, 2017-03-07:
In the past, I have used the opentx 2.2 rc10 without any problems.
Then what firmware should I put ???

3djc, 2017-03-07:
One **without** EU option choosen

wolflonezx11, 2017-03-07:
Eu became the choice
It was my fault.
sorry
Troubleshooting